ngoquang2708/android_device_samsung_vivalto3gvn #5: Camera not work with new source code

ngoquang2708, 2016-02-09:


koquantam, 2016-04-01:
Please close this, as it's working now

ngoquang2708, 2016-04-02:
Did you test it? I means compile the source codes, not use the blobs.

koquantam, 2016-04-02:
Of course, because stock blobs causes "Cannot connect to camera" bug

ngoquang2708, 2016-04-02:
I don't know what you means. We ARE using stock blobs. I disabled the camera source already.

koquantam, 2016-04-02:
Umm... in CM12.1 bro

koquantam, 2016-04-02:
Ok, i think again
When compiling, i forgot to comment out the stock blob, so still unconfirmed now :D

ngoquang2708, 2016-04-02:
These lines
```
# Camera
PRODUCT_PACKAGES += \
	camera.scx15 \
	camera2.scx15
```
in ```device_vivalto3gvn.mk``` do nothing right now!

koquantam, 2016-04-02:
Lots of compile error bro

koquantam, 2016-04-02:
A real pain in the 4$$ fixing that

ngoquang2708, 2016-04-02:
Living with blobs for now. Trying fixing video.

koquantam, 2016-04-02:
Btw, going to compile CM13
Not sure i have enough RAM or not to build the ROM with prebuilt chromium

ngoquang2708, 2016-04-02:
Search around to find someone compiled it :D

koquantam, 2016-04-02:
Yeah, the prebuilt chromium

koquantam, 2016-07-31:
I think, our sensor codes aren't in the camera source

closed 2018-06-05 by ngoquang2708