ngoquang2708/android_device_samsung_vivalto3gvn #9: error gralloc

cleverior, 2016-02-23:
i get another error..

> target thumb C++: gralloc.scx15 <= device/samsung/vivalto3gvn/libs/gralloc/gralloc_module.cpp
> target thumb C++: gralloc.scx15 <= device/samsung/vivalto3gvn/libs/gralloc/alloc_device.cpp
> target thumb C++: gralloc.scx15 <= device/samsung/vivalto3gvn/libs/gralloc/framebuffer_device.cpp
> In file included from device/samsung/vivalto3gvn/libs/gralloc/alloc_device.cpp:42:0:
> system/core/libion/include/ion/ion.h:34:35: error: 'ion_user_handle_t' has not been declared
>                unsigned int flags, ion_user_handle_t *handle);
>                                    ^
> system/core/libion/include/ion/ion.h:38:22: error: 'ion_user_handle_t' has not been declared
>  int ion_free(int fd, ion_user_handle_t handle);
>                       ^
> system/core/libion/include/ion/ion.h:39:21: error: 'ion_user_handle_t' has not been declared
>  int ion_map(int fd, ion_user_handle_t handle, size_t length, int prot,
>                      ^
> system/core/libion/include/ion/ion.h:41:23: error: 'ion_user_handle_t' has not been declared
>  int ion_share(int fd, ion_user_handle_t handle, int *share_fd);
>                        ^
> system/core/libion/include/ion/ion.h:42:38: error: 'ion_user_handle_t' has not been declared
>  int ion_import(int fd, int share_fd, ion_user_handle_t *handle);
>                                       ^
> device/samsung/vivalto3gvn/libs/gralloc/alloc_device.cpp: In function 'int gralloc_alloc_buffer(alloc_device_t*, size_t, int, const native_handle_t**)':
> device/samsung/vivalto3gvn/libs/gralloc/alloc_device.cpp:258:22: error: 'GRALLOC_USAGE_VIDEO_BUFFER' was not declared in this scope
>          if (usage & (GRALLOC_USAGE_VIDEO_BUFFER|GRALLOC_USAGE_CAMERA_BUFFER))
>                       ^
> device/samsung/vivalto3gvn/libs/gralloc/alloc_device.cpp:258:49: error: 'GRALLOC_USAGE_CAMERA_BUFFER' was not declared in this scope
>          if (usage & (GRALLOC_USAGE_VIDEO_BUFFER|GRALLOC_USAGE_CAMERA_BUFFER))
>                                                  ^
> device/samsung/vivalto3gvn/libs/gralloc/alloc_device.cpp:262:25: error: 'GRALLOC_USAGE_OVERLAY_BUFFER' was not declared in this scope
>          else if(usage & GRALLOC_USAGE_OVERLAY_BUFFER)
>                          ^
> device/samsung/vivalto3gvn/libs/gralloc/alloc_device.cpp:276:82: error: cannot convert 'ion_handle**' to 'int*' for argument '6' to 'int ion_alloc(int, size_t, size_t, unsigned int, unsigned int, int*)'
>          ret = ion_alloc(m->ion_client, size, 0, ion_heap_mask, ion_flag, &ion_hnd);
>                                                                                   ^
> device/samsung/vivalto3gvn/libs/gralloc/alloc_device.cpp:283:61: error: invalid conversion from 'ion_handle*' to 'int' [-fpermissive]
>          ret = ion_share( m->ion_client, ion_hnd, &shared_fd );
>                                                              ^
> In file included from device/samsung/vivalto3gvn/libs/gralloc/alloc_device.cpp:42:0:
> system/core/libion/include/ion/ion.h:41:5: error:   initializing argument 2 of 'int ion_share(int, int, int*)' [-fpermissive]
>  int ion_share(int fd, ion_user_handle_t handle, int *share_fd);
>      ^
> device/samsung/vivalto3gvn/libs/gralloc/alloc_device.cpp:287:56: error: invalid conversion from 'ion_handle*' to 'int' [-fpermissive]
>              if ( 0 != ion_free( m->ion_client, ion_hnd ) ) AERR( "ion_free( %d ) failed", m->ion_client );
>                                                         ^
> In file included from device/samsung/vivalto3gvn/libs/gralloc/alloc_device.cpp:42:0:
> system/core/libion/include/ion/ion.h:38:5: error:   initializing argument 2 of 'int ion_free(int, int)' [-fpermissive]
>  int ion_free(int fd, ion_user_handle_t handle);
>      ^
> device/samsung/vivalto3gvn/libs/gralloc/alloc_device.cpp:295:56: error: invalid conversion from 'ion_handle*' to 'int' [-fpermissive]
>              if ( 0 != ion_free( m->ion_client, ion_hnd ) ) AERR( "ion_free( %d ) failed", m->ion_client );
>                                                         ^
> In file included from device/samsung/vivalto3gvn/libs/gralloc/alloc_device.cpp:42:0:
> system/core/libion/include/ion/ion.h:38:5: error:   initializing argument 2 of 'int ion_free(int, int)' [-fpermissive]
>  int ion_free(int fd, ion_user_handle_t handle);
>      ^
> device/samsung/vivalto3gvn/libs/gralloc/alloc_device.cpp:313:60: error: 'ion_invalidate_fd' was not declared in this scope
>              ion_invalidate_fd(hnd->ion_client,hnd->share_fd);
>                                                             ^
> device/samsung/vivalto3gvn/libs/gralloc/alloc_device.cpp:322:52: error: invalid conversion from 'ion_handle*' to 'int' [-fpermissive]
>              ret = ion_free( m->ion_client, ion_hnd );
>                                                     ^
> In file included from device/samsung/vivalto3gvn/libs/gralloc/alloc_device.cpp:42:0:
> system/core/libion/include/ion/ion.h:38:5: error:   initializing argument 2 of 'int ion_free(int, int)' [-fpermissive]
>  int ion_free(int fd, ion_user_handle_t handle);
>      ^
> device/samsung/vivalto3gvn/libs/gralloc/alloc_device.cpp: In function 'int alloc_device_alloc(alloc_device_t*, int, int, int, int, const native_handle_t**, int*)':
> device/samsung/vivalto3gvn/libs/gralloc/alloc_device.cpp:508:16: error: 'HAL_PIXEL_FORMAT_YCbCr_420_SP' was not declared in this scope
>   if (format == HAL_PIXEL_FORMAT_YCbCr_420_SP || format == HAL_PIXEL_FORMAT_YCrCb_420_SP || format == HAL_PIXEL_FORMAT_YV12 )
>                 ^
> device/samsung/vivalto3gvn/libs/gralloc/alloc_device.cpp:512:8: error: the value of 'HAL_PIXEL_FORMAT_YCbCr_420_SP' is not usable in a constant expression
[... 64 lines cut ...]
> device/samsung/vivalto3gvn/libs/gralloc/gralloc_module.cpp:345:12: warning: unused parameter 'w' [-Wunused-parameter]
> device/samsung/vivalto3gvn/libs/gralloc/gralloc_module.cpp:345:12: warning: unused parameter 'h' [-Wunused-parameter]
> device/samsung/vivalto3gvn/libs/gralloc/gralloc_module.cpp:391:12: warning: unused parameter 'module' [-Wunused-parameter]
>  static int gralloc_unlock(gralloc_module_t const* module, buffer_handle_t handle)
>             ^
> device/samsung/vivalto3gvn/libs/gralloc/gralloc_module.cpp:433:5: warning: unused parameter 'module' [-Wunused-parameter]
>  int gralloc_perform(struct gralloc_module_t const* module,
>      ^
> make: *** [/media/cleverior/Lab-Room/Other/rom/rrlolli5.1/out/target/product/vivalto3gvn/obj/SHARED_LIBRARIES/gralloc.scx15_intermediates/gralloc_module.o] Error 1
> device/samsung/vivalto3gvn/libs/gralloc/framebuffer_device.cpp: In function 'void writeFpsToProc(float)':
> device/samsung/vivalto3gvn/libs/gralloc/framebuffer_device.cpp:174:22: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
>      char *fps_proc = "/proc/benchMark/fps";
>                       ^
> device/samsung/vivalto3gvn/libs/gralloc/framebuffer_device.cpp: In function 'bool getApctFpsSupport()':
> device/samsung/vivalto3gvn/libs/gralloc/framebuffer_device.cpp:207:23: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
>      char *FILE_NAME = "/data/data/com.sprd.APCT/apct/apct_support";
>                        ^
> device/samsung/vivalto3gvn/libs/gralloc/framebuffer_device.cpp: At global scope:
> device/samsung/vivalto3gvn/libs/gralloc/framebuffer_device.cpp:732:5: warning: unused parameter 'dev' [-Wunused-parameter]
>  int compositionComplete(struct framebuffer_device_t *dev)
>      ^
> device/samsung/vivalto3gvn/libs/gralloc/framebuffer_device.cpp:751:5: warning: unused parameter 'name' [-Wunused-parameter]
>  int framebuffer_device_open(hw_module_t const *module, const char *name, hw_device_t **device)
>      ^
>

ngoquang2708, 2016-02-23:
Checkout my commits from my android_system_core and android_frameworks_native

ngoquang2708, 2016-02-23:
And android_hardware_libhardware, I forgot.

ngoquang2708, 2016-02-28:
@cleverior Do you got any other errors?

cleverior, 2016-03-02:
Nothing bro. Thanks.

closed 2016-03-02 by cleverior