ngoquang2708/android_device_samsung_vivalto3gvn #11: cm-12.1: 2nd SIM not work

ngoquang2708, 2016-02-24:
When enable DSDS ril mode, both SIMs die. Remove it, 1st SIM works fine.

koquantam, 2016-02-27:
Actually it still works, but for some reason, RIL crashes when enabling mobile data

ngoquang2708, 2016-02-27:
I remove dsds to use single sim and it works. Call, ussd...

ngoquang2708, 2016-02-28:
@koquantam I tried remove ril class from ```system.prop``` and ```BoardConfig.mk``` but 2nd SIM still not work.

ngoquang2708, 2016-02-28:
Haha, got a fix, need to confirm it myself :)

koquantam, 2016-02-28:
RIL class is mandatory tho, removing it give you no access to RIL

koquantam, 2016-02-28:
I mean calls, SMS, mobile data, etc

koquantam, 2016-03-06:
What's your solution about RIL?

ngoquang2708, 2016-03-06:
You already known it, changing ril socket names. Got 2 signal now, no mobile data, same imei for both sim.

koquantam, 2016-03-06:
IDK why those weird things happen with you. I still get both IMEIs and signals in proper way, but no mobile data only (chosen RIL crashes when enable mobile data for chosen SIM)

koquantam, 2016-03-06:
By seeing logcat, there're such things like "accept_ra_rt_table" and "use_optimistic" thingies. Just wondering what are they

ngoquang2708, 2016-03-06:
Confirmed that RIL crashes when enable mobile data for chosen SIM.

ngoquang2708, 2016-03-13:
Fixed by c5ac459449fcc1ff67df606124d9453fa243928e.

closed 2016-03-13 by ngoquang2708

koquantam, 2016-03-13:
Please rename the issue and re-open it

ngoquang2708, 2016-03-13:
What is the name it should be and why re-open it?

koquantam, 2016-03-13:
Hmm... 1st mobile data doesn't work, then re-open it as a new issue

ngoquang2708, 2016-03-13:
We are tracking it in #25, aren't we?

koquantam, 2016-03-13:
Uhh... ok, my mistake :/