ngoquang2708/android_device_samsung_vivalto3gvn #14: cm-12.1: Audio FX has no effect

ngoquang2708, 2016-02-24:


ngoquang2708, 2016-02-24:
NuPlayer again.

ngoquang2708, 2016-02-25:
Everything will be working normally when switch to AwesomePlayer in DevSettings.

koquantam, 2016-02-28:
Did your phone freezes when pressing volume buttons?

koquantam, 2016-02-28:
Not exactly freezes, but Trebuchet freezes, and after a couple of minutes, it soft-reboot

ngoquang2708, 2016-02-28:
It is working normally.

koquantam, 2016-02-28:
After turning to AwesomePlayer?

ngoquang2708, 2016-02-28:
Both NP and AP.

ngoquang2708, 2016-02-28:
Any logcat?

koquantam, 2016-02-28:
I can't get it now, but it shows that "service media.audio_flinger died" then soft-reboot
Pressing volume buttons makes Trebuchet freezes. Audio settings in Settings does the same thing, can't change audio volume

koquantam, 2016-02-28:
Btw, can i try your build?

ngoquang2708, 2016-02-28:
I am building it currently with major bug fix with libatchannel which fix audio routing to ext speaker when incall. you can try my old build [cm-12.1-20160226-UNOFFICIAL-vivalto3gvn.zip](http://qrpi.noip.me:10080/upload/cm-12.1-20160226-UNOFFICIAL-vivalto3gvn.zip)

ngoquang2708, 2016-02-28:
About crashing mediaserver, I didn't remember exactly, when change it user to ```root``` instead of ```media``` in ```init.rc```, crashing is gone. Our audio HAL require more privilege than standard one.

ngoquang2708, 2016-02-28:
May be ```system``` user should be enough but I haven't tried it.

koquantam, 2016-02-28:
We need that on CM12.1. Stock APM in CM11 makes UI audio very quiet, but media or ringtone works fine

ngoquang2708, 2016-02-28:
OK, I remember it now. I removed ```nvram ccci``` groups plus set user to ```root``` and ```mediaserver``` audio service stop crashing.

ngoquang2708, 2016-02-28:
I don't know why but audio in cm12 is perfectly, not very lough like in cm11, no audio blobs here.

ngoquang2708, 2016-02-28:
@koquantam I remembered that your cm12.1 build has both sim work, right?

koquantam, 2016-02-28:
Only calls, SMS. Mobile data works only for one SIM

ngoquang2708, 2016-02-28:
But you could see 2 network signals?

koquantam, 2016-02-28:
Yes and 2 exclamation marks on both SIMs, which means no mobile data for almost all of the time. If enabling mobile while in dual-SIM mode, that chosen SIM RIL would crash

ngoquang2708, 2016-02-28:
@koquantam Oh god, at least dual-SIM work for you, never work for me. When I enable ```dsds```, both network signal is gone but 1st SIM is usable. Could you try to flash my new build to see if ```dsds``` work (please re-enable it from ```build.prop```).

@Y300-0100 If you cannot wait, try this.

Here is it: [cm-12.1-20160228-UNOFFICIAL-vivalto3gvn.zip](http://qrpi.noip.me:10080/upload/cm-12.1-20160228-UNOFFICIAL-vivalto3gvn.zip)

koquantam, 2016-02-28:
@ngoquang2708 
vivalto3gvn build cannot work on kanas, as they had different hardware

ngoquang2708, 2016-02-28:
I though he had galaxy v.

koquantam, 2016-02-28:
He has mentioned that he was working for G355HN, Galaxy Core 2 with NFC

koquantam, 2016-02-28:
Did you integrated libmemoryheapion?

ngoquang2708, 2016-02-28:
Not yet. Using ```LD_PRELOAD``` is just a hack, much like we are hacking ```libbinder``` right now. If we can make camera codes to work, then we can use ```libmemoryheapion``` without ```LD_PRELOAD``` hacking and ```libbinder``` hacking too.

koquantam, 2016-02-28:
Bro, not just camera relies on ION. Anything on HW needs ION too, so just make it temporary hack instead. People want a lollipop build instead of a bugless build

ngoquang2708, 2016-02-28:
Of course, when camera source codes work, we can change hwc source code or any other source code to use ```libmemoryheapion```!

ngoquang2708, 2016-02-28:
And the current hack on libbinder is working, bro!

koquantam, 2016-02-28:
Yes, it's working
Dual-SIM seems broken. Could you try removing RIL CFLAGS?
More weird is: SIM1 can call, but SIM2 can't. Starting to have some headaches

ngoquang2708, 2016-02-28:
You said Dual-SIM is broken, so SIM2 supported to not work?!?!?

koquantam, 2016-02-28:
"supposed" , not "supported" dude
Yes, i've some typos :D

ngoquang2708, 2016-02-28:
Yes, my grammar it not good. Thanks for pointing it out any way :D

closed 2016-02-28 by ngoquang2708