ngoquang2708/android_device_samsung_vivalto3gvn #15: cm-12.1: YouTube app cannot play video

ngoquang2708, 2016-02-25:
It just produce some sounds but the video won't move.

thanhthuy89, 2016-03-10:
edit buildprop 5.1.1 to 4.4.2 or 4.4.4

ngoquang2708, 2016-03-10:
@thanhthuy89  I will try.

thanhthuy89, 2016-03-10:
when I delete fonts from theme by root explorer that it will does not have hanging logo status

thanhthuy89, 2016-03-10:
bug from font

ngoquang2708, 2016-03-14:
YouTube crashes mediaserver when it try to load DRM plugins.

ngoquang2708, 2016-03-14:
And DRM plugins is failed to load in CM12.1. How on CM11.0?

ngoquang2708, 2016-03-22:
OMG I can play 144p video :))

ngoquang2708, 2016-03-22:
ZingMp3 can play 720p =))

ngoquang2708, 2016-03-28:
@koquantam Did you try to plauy video in ZingMP3? Why Zing MP3 can play videos, even 720p, while YT cannot :(

koquantam, 2016-03-28:
The strange guy on XDA said that he disabled HW rendering and YT works. I haven't tested it yet

ngoquang2708, 2016-04-08:
Fixed by a093f22e2aa7207b756afee27989ea6dfbea4b5d.

closed 2016-04-08 by ngoquang2708