ngoquang2708/android_device_samsung_vivalto3gvn #16: cm-12.1: camera is crashed when switch from recording mode to capture mode

ngoquang2708, 2016-02-25:
After that, Torch stops working.

koquantam, 2016-02-26:
How the hell you got camera working?

koquantam, 2016-02-26:
I'm pretty damn crazy with these stuffs in CM12.1

ngoquang2708, 2016-02-27:
Use these [patches](https://github.com/ngoquang2708/android_device_samsung_vivalto3gvn/tree/cm-11.0/patches) and my cm-12.1 branch.

ngoquang2708, 2016-02-27:
@koquantam Why don't you make cm-12.1 branch from cm-11.0 branch so that I could send pull requests.

koquantam, 2016-02-27:
Well, i want to get rid of those source building, since my machine is not so powerful as your. CM12.1 takes so damn long even with prebuilt chromium because of my dual-core CPU

ngoquang2708, 2016-02-27:
What do you means ```those source building```? Didn't you just build it? Building cm-12.1 is nothing to do with git history, I just want to keep it so that I and you have the same git history, and sync our changes.

koquantam, 2016-02-27:
Oops, sorry for typos :D
I mean that the vendor sources (libs in our device repo). About changes, should we make a team? It's better than sending pull requests

ngoquang2708, 2016-02-27:
OK, what will be the name? I cannot think out any name :D

koquantam, 2016-02-28:
Build 20160228 doesn't exist anymore

ngoquang2708, 2016-02-28:
Maybe this commit 6dea3064fd7403875223a97226e317b0a842739b solves it.

closed 2016-02-28 by ngoquang2708