ngoquang2708/android_device_samsung_vivalto3gvn #20: cm-12.1: Bluetooth FC when sending and receiving data

koquantam, 2016-03-03:
I don't have a phone to test this for now, but a person reported that BT acts like that

ngoquang2708, 2016-03-03:
Ok, I will take a look.

ngoquang2708, 2016-03-03:
logcat:
```
W/bt-btm  ( 1363): btm_acl_role_changed: BDA: a4-db-30-d6-71-42
W/bt-btm  ( 1363): btm_acl_role_changed: New role: 0
W/bt-btm  ( 1363): btm_acl_created hci_handle=11 link_role=1  transport=1
W/bt-btm  ( 1363): btm_acl_created hci_handle=11 link_role=0  transport=1
W/bt-btm  ( 1363): btm_read_remote_version_complete: BDA: a4-db-30-d6-71-42
W/bt-btm  ( 1363): btm_read_remote_version_complete lmp_version 7 manufacturer 69 lmp_subversion 1
W/bt-btm  ( 1363): btm_process_remote_ext_features_page 0: BDA: a4-db-30-d6-71-42
W/bt-btm  ( 1363): ext_features_complt page_num:1 f[0]:x03, sm4:11, pend:0
W/bt-btm  ( 1363): btm_process_remote_ext_features_page 1: BDA: a4-db-30-d6-71-42
W/bt-btif ( 1363): info:x10
D/        ( 1363): remote version info [a4:db:30:d6:71:42]: 7, 45, 1
W/bt-l2cap( 1363): L2CA_SetDesireRole() new:x0, disallow_switch:0
W/bt-l2cap( 1363): L2CAP - st: CLOSED evt: 10
W/bt-l2cap( 1363): L2CAP - st: TERM_W4_SEC_COMP evt: 7
W/bt-l2cap( 1363): L2CA_ErtmConnectRsp()  CID: 0x0040  Result: 0  Status: 0  BDA: a4db30d67142  p_ertm_info:0x00000000
W/bt-l2cap( 1363): L2CAP - st: W4_L2CA_CON_RSP evt: 22
W/bt-l2cap( 1363): L2CAP - st: CONFIG evt: 24
W/bt-l2cap( 1363): L2CAP - st: CONFIG evt: 14
W/bt-l2cap( 1363): L2CAP - st: CONFIG evt: 25
W/bt-l2cap( 1363): L2CAP-Upper layer Config_Rsp,Local CID: 0x0040,Remote CID: 0x0040,PSM: 1,our MTU present:1,our MTU:672
W/bt-l2cap( 1363): L2CAP - st: CONFIG evt: 15
W/bt-l2cap( 1363): L2CAP-peer_Config_Rsp,Local CID: 0x0040,Remote CID: 0x0040,PSM: 1,peer MTU present: 0,peer MTU: 672
W/bt-l2cap( 1363): L2CA_DisconnectRsp()  CID: 0x0040
W/bt-l2cap( 1363): L2CAP - st: W4_L2CA_DISC_RSP evt: 28
W/bt-btm  ( 1363): Security Manager: encrypt_change status:0 State:0, encr_enable = 1
W/bt-l2cap( 1363): L2CAP - st: CLOSED evt: 10
W/bt-l2cap( 1363): L2CAP - st: TERM_W4_SEC_COMP evt: 7
W/bt-l2cap( 1363): SOCK_L2C_ConnectInd LCID:0x41
W/bt-l2cap( 1363): L2CA_ErtmConnectRsp()  CID: 0x0041  Result: 0  Status: 0  BDA: a4db30d67142  p_ertm_info:0xa5d07a71
W/bt-l2cap( 1363): L2CAP - st: W4_L2CA_CON_RSP evt: 22
W/bt-l2cap( 1363): L2CAP - st: CONFIG evt: 24
W/bt-l2cap( 1363): L2CAP - st: CONFIG evt: 14
W/bt-l2cap( 1363): L2CAP - st: CONFIG evt: 25
W/bt-l2cap( 1363): L2CAP-Upper layer Config_Rsp,Local CID: 0x0041,Remote CID: 0x0041,PSM: 5255,our MTU present:1,our MTU:65000
W/bt-l2cap( 1363): L2CAP - st: CONFIG evt: 15
W/bt-l2cap( 1363): L2CAP-peer_Config_Rsp,Local CID: 0x0041,Remote CID: 0x0041,PSM: 5255,peer MTU present: 0,peer MTU: 32767
W/bt-l2cap( 1363): L2CA_SetDesireRole() new:x0, disallow_switch:0
W/bt-btif ( 1363): new conn_srvc id:26, app_id:255
W/bt-btif ( 1363): new conn_srvc id:26, app_id:255 count:1
W/bt-btif ( 1363): bta_dm_pm_ssr conn_srvc id:26, app_id:255
W/bt-btif ( 1363): bta_dm_pm_ssr:2, lat:1200
I/BtOppL2CapListener( 1363): Accepted connection from A4:DB:30:D6:71:42
I/art     ( 1363): Rejecting re-init on previously-failed class java.lang.Class<com.android.bluetooth.opp.BluetoothOppTransport>
--------- beginning of crash
E/AndroidRuntime( 1363): FATAL EXCEPTION: BtOppL2CapListener
E/AndroidRuntime( 1363): Process: com.android.bluetooth, PID: 1363
E/AndroidRuntime( 1363): java.lang.NoClassDefFoundError: com.android.bluetooth.opp.BluetoothOppTransport
E/AndroidRuntime( 1363): 	at com.android.bluetooth.opp.BluetoothOppL2capListener$1.run(BluetoothOppL2capListener.java:168)
D/OpenGLRenderer(  515): Use EGL_SWAP_BEHAVIOR_PRESERVED: false
D/Atlas   (  515): Validating map...
I/OpenGLRenderer(  515): Initialized EGL, version 1.4
D/OpenGLRenderer(  515): Enabling debug mode 0
D/gralloc.scx15(  124): alloc_device_alloc w:576, h:328, format:1 usage:0xf02 start
D/gralloc.scx15(  124): the flag 0x4 and the vadress is 0xb4896000 and the size is 0xb8800
D/gralloc.scx15(  124): alloc_device_alloc handle:0xb656f820 end err is 0
D/gralloc.scx15(  124): alloc_device_alloc w:576, h:328, format:1 usage:0xf02 start
D/gralloc.scx15(  124): the flag 0x4 and the vadress is 0xb471f000 and the size is 0xb8800
D/gralloc.scx15(  124): alloc_device_alloc handle:0xb6473f40 end err is 0
D/gralloc.scx15(  124): alloc_device_alloc w:576, h:328, format:1 usage:0xf02 start
D/gralloc.scx15(  124): the flag 0x4 and the vadress is 0xb4666000 and the size is 0xb8800
D/gralloc.scx15(  124): alloc_device_alloc handle:0xb656f1c0 end err is 0
D/gralloc.scx15(  124): alloc_device_alloc w:576, h:328, format:1 usage:0xf02 start
D/gralloc.scx15(  124): the flag 0x4 and the vadress is 0xb45ad000 and the size is 0xb8800
D/gralloc.scx15(  124): alloc_device_alloc handle:0xb656f0a0 end err is 0
E/        ( 1631): main: slog begin to work. [12]
E/        ( 1631): handle_slog_property: slog.step = 0, slog.reload = 0 [12]
E/        ( 1631): parse_config: cannot find config files! [2]
```

ngoquang2708, 2016-03-03:
Look like ```Bluetooth.apk``` cannot access ```/system/framework/javax.btobex.jar``` library.

ngoquang2708, 2016-03-06:
Got a fix. I ```/system/etc/permissions/platform.xml``` add this
```
    <library name="javax.btobex"
        file="/system/framework/javax.btobex.jar" />
```
before
```
</permissions>
```

ngoquang2708, 2016-03-06:
Factory reset if needed.

ngoquang2708, 2016-03-07:
Fixed by d25c8526d58c23fa20cc199e5baec90ddf595d1a.

closed 2016-03-07 by ngoquang2708

koquantam, 2016-03-07:
I'm sorry, but can you work with CM12.1 for now? Because i'm stuck with exams, and about 2 months i get out school, so i can continue working with this device. Now the least i can do is working with custom kernel

koquantam, 2016-03-07:
When you feel it stable enough, you can release the ROM. You are a contributor, so you can easily edit the OP to update the latest builds

ngoquang2708, 2016-03-07:
OK.

thanhthuy89, 2016-03-09:
bạn ơi bạn đã fix duoc bluethooth buộc dừng khi gởi và nhận dữ liệu chưa , với áp dụng giao diện mà tick vao "font" se bootloop chưa?? up len cho ae tham khao đi.
minh flash cm12 vô sua buildprop xuong 4.4.2 va xoa dong low_ram_config gì đó la vô youtobe coi bt. con về video tải về để xem thì dùng VLC beta tren chplay la coi duoc. rieng quay video thi mac dinh e nó co 3mgx thì quay xấu thấy bà... e dung phan mềm seltfile camera tren chplay xài, nó rút gọn ko còn chức năng video. Mình có nhiêu ý đó. Mong bạn sớm update cho ae cung thưởng thức với. Đang mong!!

koquantam, 2016-03-10:
@thanhthuy89 đã fix và bác quang sẽ up bản mới
Tiện thể, đây là trên Git, bạn vui lòng nói = tiếng anh để khi người khác vào xem mã nguồn, họ hiểu được những gì mình nói

thanhthuy89, 2016-03-10:
release cm12 
how long?

ngoquang2708, 2016-03-10:
@thanhthuy89 It will be released soon, but if you cannot wait, this ROM is not for you. I'm sorry.