ngoquang2708/android_device_samsung_vivalto3gvn #23: CM11 : error compile SamsungSPRDRIL.java

cleverior, 2016-03-03:
here the logcat

> frameworks/opt/telephony/../../../device/samsung/vivalto3gvn/ril/com/android/internal/telephony/SamsungSPRDRIL.java:45: error: constructor RIL in class RIL cannot be applied to given types;
>         super(context, networkMode, cdmaSubscription, instanceId);
>         ^
>   required: Context,int,int
>   found: Context,int,int,Integer
>   reason: actual and formal argument lists differ in length
> frameworks/opt/telephony/../../../device/samsung/vivalto3gvn/ril/com/android/internal/telephony/SamsungSPRDRIL.java:85: error: cannot find symbol
>             if (mSubscriptionStatusRegistrants != null) {
>                 ^
>   symbol:   variable mSubscriptionStatusRegistrants
>   location: class SamsungSPRDRIL
> frameworks/opt/telephony/../../../device/samsung/vivalto3gvn/ril/com/android/internal/telephony/SamsungSPRDRIL.java:86: error: cannot find symbol
>                 mSubscriptionStatusRegistrants.notifyRegistrants(
>                 ^
>   symbol:   variable mSubscriptionStatusRegistrants
>   location: class SamsungSPRDRIL
> frameworks/opt/telephony/../../../device/samsung/vivalto3gvn/ril/com/android/internal/telephony/SamsungSPRDRIL.java:91: error: cannot find symbol
>             if (mSubscriptionStatusRegistrants != null) {
>                 ^
>   symbol:   variable mSubscriptionStatusRegistrants
>   location: class SamsungSPRDRIL
> frameworks/opt/telephony/../../../device/samsung/vivalto3gvn/ril/com/android/internal/telephony/SamsungSPRDRIL.java:92: error: cannot find symbol
>                 mSubscriptionStatusRegistrants.notifyRegistrants(
>                 ^
>   symbol:   variable mSubscriptionStatusRegistrants
>   location: class SamsungSPRDRIL
> frameworks/opt/telephony/../../../device/samsung/vivalto3gvn/ril/com/android/internal/telephony/SamsungSPRDRIL.java:99: error: cannot find symbol
>         int simId = mInstanceId == null ? 0 : mInstanceId;
>                     ^
>   symbol:   variable mInstanceId
>   location: class SamsungSPRDRIL
> frameworks/opt/telephony/../../../device/samsung/vivalto3gvn/ril/com/android/internal/telephony/SamsungSPRDRIL.java:99: error: cannot find symbol
>         int simId = mInstanceId == null ? 0 : mInstanceId;
>                                               ^
>   symbol:   variable mInstanceId
>   location: class SamsungSPRDRIL
> frameworks/opt/telephony/../../../device/samsung/vivalto3gvn/ril/com/android/internal/telephony/SamsungSPRDRIL.java:114: error: cannot find symbol
>             if (mInstanceId != null) {
>                 ^
>   symbol:   variable mInstanceId
>   location: class SamsungSPRDRIL
> Note: Some input files use or override a deprecated API.
> Note: Recompile with -Xlint:deprecation for details.
> Note: Some input files use unchecked or unsafe operations.
> Note: Recompile with -Xlint:unchecked for details.
> 8 errors

ngoquang2708, 2016-03-04:
Please compare your ril class with mine.

ngoquang2708, 2016-03-04:
Why don't you just fork my repo or @koquantam repo, instead of doing your
own repo and report errors here?

cleverior, 2016-03-10:
which repo ? i've replace all of your repo and from koquantam. but still get error when compiling ril. i dont know why. maybe this is bcz im using source CM from high compresed without .repo. So, my source is out of date.

ngoquang2708, 2016-03-10:
I can compress my CM sources and send you if you wish.

ngoquang2708, 2016-03-10:
I've checked. Your source tree is out of date compare to my device source tree.

cleverior, 2016-03-10:
Yes if you dont mind. Btw, have you check your email and PM xda ? i've sent invitation to your email.

ngoquang2708, 2016-03-10:
Checked. Accepted :D

ngoquang2708, 2016-03-10:
Use this 
[cm-11.0.7z.torrent.zip](https://github.com/ngoquang2708/android_device_samsung_vivalto3gvn/files/167298/cm-11.0.7z.torrent.zip) to download cm-11.0 source. I am seeding it from my tiny home server (R.Pi).

ngoquang2708, 2016-03-11:
I can see that download speed is slow, can you try [this](http://qrpi.noip.me:10080/upload/cm-11.0.7z) to see if it is faster.

ngoquang2708, 2016-03-11:
I have a question, I am already a member vivalto group, so do I have write permissions to its repositories?

koquantam, 2016-03-11:
Yes, you do
I can fork my project there

ngoquang2708, 2016-03-11:
Still no write access.

ngoquang2708, 2016-03-11:
OK, found it, I just can send PR https://help.github.com/articles/permission-levels-for-an-organization/

cleverior, 2016-03-14:
thanks bro. direct link is better than torrent for me. :D

ngoquang2708, 2016-03-14:
I though they're the same. Both from my tiny server!

ngoquang2708, 2016-03-14:
You can remove the kernel and the device tree, then clone git repo to get update.

cleverior, 2016-03-15:
ok. could u compress your cm12 source also ? thanks in advance.

ngoquang2708, 2016-03-15:
Yes

ngoquang2708, 2016-03-15:
@cleverior Again, 2 methods to download:

- [Torrent](https://github.com/ngoquang2708/android_device_samsung_vivalto3gvn/files/173781/cm-12.1.7z.torrent.zip)

- [Direct](http://qrpi.noip.me:10080/upload/cm-12.1.7z)

cleverior, 2016-03-16:
@ngoquang2708 i cant download from direct link bro. could you check your mini server ?

ngoquang2708, 2016-03-16:
It is operating normally. What's error? Please check again.

ngoquang2708, 2016-03-16:
I recommend torrent. It is slow at first, but it is safer with more integrity checking.

cleverior, 2016-03-16:
it cant start download. even using IDM.

ngoquang2708, 2016-03-16:
IDM will kill my server! Use browser please!

cleverior, 2016-03-16:
and there is error "Cannot download the file because of timeout. Cannot connect to qrpi.noip.me:10080"

ngoquang2708, 2016-03-16:
OK, use my IP instead: 14.187.49.86

ngoquang2708, 2016-03-16:
You can ping qrpi.noip.me to see if it is the same like my IP :D

cleverior, 2016-03-16:
username and passwd ?

ngoquang2708, 2016-03-16:
You know, cheap internet sub w/ dynamic IP :D

ngoquang2708, 2016-03-16:
What username???

ngoquang2708, 2016-03-16:
http://14.187.49.86:10080/upload/cm-12.1.7z

ngoquang2708, 2016-03-16:
That IP will change.

cleverior, 2016-03-16:
This site can’t be reached
14.187.49.86 took too long to respond.
ERR_CONNECTION_TIMED_OUT

:(

ngoquang2708, 2016-03-16:
What about this: http://14.187.49.86:10080/

cleverior, 2016-03-16:
still get the same error

ngoquang2708, 2016-03-16:
Aghh, Firewall???????????????
Could you try torrent?

cleverior, 2016-03-16:
still "connecting to peers 0.0%" since yesterday

cleverior, 2016-03-16:
i've disable my firewall

ngoquang2708, 2016-03-16:
OK, I will try another HTTP port :(

ngoquang2708, 2016-03-16:
What is the ip when u ping qrpi.noip.me?

cleverior, 2016-03-16:
thanks in advance. sorry if i disturbing your time.

cleverior, 2016-03-16:
14.187.49.86

ngoquang2708, 2016-03-16:
OK, try to connect to it through port 54455

ngoquang2708, 2016-03-16:
Any ideas about file sharing services???

cleverior, 2016-03-16:
still cant connect. could you upload it to google drive ?

ngoquang2708, 2016-03-16:
OK.

ngoquang2708, 2016-03-16:
I will send you the link about 2 hours later.

cleverior, 2016-03-16:
OK. Thanks again.

ngoquang2708, 2016-03-16:
I bet ISP block your incoming connection :D

koquantam, 2016-03-16:
You know, his country is very far from our city, so it can't reach us. Also you're using mobile ISP, which has limited connection, so how can he download your source properly?

ngoquang2708, 2016-03-16:
I use FTTH, up/down -> 20Mbps/20Mbps, not bad!

cleverior, 2016-03-16:
I can download cm11 from direct link that shared by ngoquang. But why i
cant do the same thing with cm12.1 link ?

ngoquang2708, 2016-03-16:
@cleverior Can you download cm-11.0 link right now :D?

ngoquang2708, 2016-03-16:
@cleverior Do you compile cm-11.0 successfully? If it is, please close this issue!

ngoquang2708, 2016-03-16:
@cleverior Link Link https://drive.google.com/file/d/0B81z8fcTnClGN2NIbVFkWVZNbkE/view?usp=sharing

cleverior, 2016-03-17:
i just download the source. not try to build yet. I'll try to build later if i have free time. Btw, thanks for help me. You can delete all the links.

closed 2016-03-17 by cleverior