ngoquang2708/android_device_samsung_vivalto3gvn #25: cm-12.1: sim1's rild is crashed when turn mobile data on

ngoquang2708, 2016-03-09:
Just like in cm-11.0.

koquantam, 2016-03-09:
Actually was both SIM RILs
The mobile data only work with single-SIM mode

thanhthuy89, 2016-03-10:
single sim

ngoquang2708, 2016-03-13:
OK, Success for turn on mobile on SIM2, still no SIM1. Here radio logcat when enable mobile data on SIM1 [cm-12.1-vivalto3gvn.sim1+mobiledata.radio.txt](https://gist.github.com/ngoquang2708/a216099b71f3fb0fc59b)

ngoquang2708, 2016-03-13:
Seem ```rild``` crashes on this ```D/use-Rlog/RLOG-RIL( 184): [RIL::FD] Load Default Policy```

koquantam, 2016-03-13:
Maybe

ngoquang2708, 2016-03-13:
Here radio logcat when success enable mobile data on SIM2 [cm-12.1-vivalto3gvn.sim2+mobiledata.radio.txt](https://gist.github.com/ngoquang2708/341a490ed3815bf121e0)

koquantam, 2016-03-13:
You can see the MediaTek RIL for some good information
https://github.com/Al3XKOoL/android_device_wiko_rainbow/blob/cm-12.1/ril/telephony/java/com/android/internal/telephony/MediaTekRIL.java#L1172

ngoquang2708, 2016-03-13:
This prop ```gsm.operator.numeric``` is killing our ```rild1```, set it to empty string before start ```rild1```, and rild will success make data connection. Can you try this on CM-11.0.

koquantam, 2016-03-13:
It's already empty on CM-11.0
[image]

ngoquang2708, 2016-03-13:
Haha, that why rild crash on cm-12.1.

koquantam, 2016-03-13:
Btw, we're going to have Linux 3.10.40 kernel soon

ngoquang2708, 2016-03-13:
Good news!

koquantam, 2016-03-13:
I'm on the first stage of upgrading it. This might takes a lot of time, so be wait
I'll include it with OC Kernel

koquantam, 2016-03-13:
Or maybe stock kernel

ngoquang2708, 2016-03-13:
It seems android 5.1 use ```,``` to concat multiple prop value for multi-sim instead of use separated name for a prop. Ex:
```
gsm.operator.numeric = [123,234]
```
instead of
```
gsm.operator.numeric = 123
gsm.operator.numeric2 = 234
```

ngoquang2708, 2016-03-13:
Did you apply this [patch](https://gist.github.com/pawitp/9433442) in cm-11.0. I can see that in your screenshot. Don't do that, our rild doesn't use that prop naming scheme.

koquantam, 2016-03-13:
Yes, i've applied

koquantam, 2016-03-13:
But that patch is useless in CM12.1, since MSIM has been integrated to telephony-common framework

ngoquang2708, 2016-03-13:
No, I haven't applied it. I am trying to patch cm12.1's frameworks/base/telephony using the prop naming scheme that stock use, hope that rild1 will not crash.

ngoquang2708, 2016-03-13:
Interesting, rild of @TeamButter's mint have the same naming scheme in @pawitp's [patch](https://gist.github.com/pawitp/9433442), but why their rild crash when change rild socket name in RIL.java http://forum.xda-developers.com/showpost.php?p=65811052&postcount=129?
RIL's socket name supposed to be [rild, rild1, rild2], and their [init.*.rc.](https://github.com/TeamButter/android_device_samsung_mint/blob/cm-11.0/rootdir/init.sp8810.rc#L375) have the same socket names.

koquantam, 2016-03-13:
What do you mean "the same"?

ngoquang2708, 2016-03-13:
```
quang@ubuntu:~/Downloads$ strings libreference-ril.so | grep gsm
gsm.operator.numeric_1
gsm.operator.numeric
gsm.sim.operator.numeric_1
gsm.sim.operator.numeric
```

ngoquang2708, 2016-03-13:
But our RIL uses this naming scheme
```
gsm.operator.numeric
gsm.operator.numeric2
```

koquantam, 2016-03-13:
It was in the shared library source, so why you modify it in framework?

ngoquang2708, 2016-03-13:
Here is the exactly file that I've modified to change the naming scheme: frameworks/base/telephony/java/android/telephony/TelephonyManager.java, look at getTelephonyProperty() and setTelephonyProperty() method.

thanhthuy89, 2016-03-14:
help :
*.Switching to other fonts will get bootloops after restart

koquantam, 2016-03-14:
@thanhthuy89 
That's common issue, no persistent solution for now

ngoquang2708, 2016-03-14:
Fixed by https://github.com/ngoquang2708/android_frameworks_base/commit/4dc3063662e7c3d969f1a7b80a6d13b85bc33c0e.

closed 2016-03-14 by ngoquang2708