ngoquang2708/android_device_samsung_vivalto3gvn #28: cm-12.1: needs a proper SELinux configuration

ngoquang2708, 2016-03-13:
Track https://github.com/koquantam/android_device_samsung_vivalto3gvn/issues/86 here.

koquantam, 2016-03-16:
Oh dear, lots of RAM. Free to play games :D
However, this is a bug of OC Kernel. Seems it exist since patching UKSM
[image]

ngoquang2708, 2016-03-16:
Haha, huge amount of RAM!

thanhthuy89, 2016-03-17:
share OC kernel !!!

koquantam, 2016-03-17:
@thanhthuy89 
On XDA, Galaxy Ace 4 kernel. I've included CM12.1 version too

koquantam, 2016-03-18:
Bro, do you know how to build custom recovery?
I need that to test f2fs with the phone's eMMC

ngoquang2708, 2016-03-18:
Replace CM recovery source code with custom recovery source code.

koquantam, 2016-03-18:
Bad news is: My old HDD was accidentally dead, which was in recovery now. I purchased a new HDD and now have to use Windows to compile kernel :D

koquantam, 2016-03-18:
We need sth in the ROM to let it mount as f2fs, at least with update-binary, since my TWRP can't install the ROM with f2fs zip

ngoquang2708, 2016-03-18:
I don't know. But I think f2fs it not good enough. I tried f2fs on my Raspberry Pi, and when it lose power, the file system is corrupted. It is not the case with ext4.

koquantam, 2016-03-18:
We use battery to power on the phone, so how can it lose power immediately? At least with insta-reboot, which makes the fs corrupted w/o fsync

koquantam, 2016-03-18:
And R.Pi uses SD to boot

koquantam, 2016-03-18:
My 16GB SD has 3GB partitioned f2fs and mkfs.f2fs has successfully mounted it, even formatting CSC partition still be able to mount and copy data there, but I don't understand why the system won't boot with f2fs data partition, at least with me

koquantam, 2016-03-20:
Now i realized that USB OTG was in the kernel source, and it was in defconfig too LOL
`CONFIG_USB_OTG=y
# CONFIG_USB_OTG_WHITELIST is not set
# CONFIG_USB_OTG_BLACKLIST_HUB is not set`

ngoquang2708, 2016-03-20:
Haha, I don't have a cable to test it :D

koquantam, 2016-03-20:
At first, it was not set, now i have enabled it
Need volunteers to test it

koquantam, 2016-03-20:
Surprise, it doesn't work
The main OTG source was removed by SPRD, and i need to port it from another kernels, which is a PITA
Nevermind, gotta do it when have time

koquantam, 2016-03-20:
`CONFIG_USB_OTG` just a trick from Linus, doesn't do anything

ngoquang2708, 2016-03-20:
:)

koquantam, 2016-03-24:
Do you know how to make AROMA Installer?
I'll get my kernels there, so no need separate boot.img. But... that seems can't be flashed

ngoquang2708, 2016-03-24:
First time heard about it

koquantam, 2016-03-24:
Oh dear

koquantam, 2016-03-24:
It allows to install different versions of ROMs or kernels, just in one zip

koquantam, 2016-03-24:
Yeah, Linux 3.10.18 :D
[image]

koquantam, 2016-03-28:
Haha, this is what we need for USB OTG. Our OC Kernel missing these, so OTG doesn't work
https://github.com/sprd-dev/kernel_common/tree/sprdroid4.4/drivers/usb/gadget/dwc_otg

ngoquang2708, 2016-03-28:
Our current kernel tree has that source. Just need some configuration in
fstab. I made some modification in fstab to support it in the future (when
I have a OTG cable :D).

koquantam, 2016-03-28:
Yes, but missing some codes, which is needed for OTG
I've compiled, but.... white stripes :D

koquantam, 2016-03-29:
2MB more RAM :D
[image]

ngoquang2708, 2016-03-29:
2G will good :D

koquantam, 2016-03-29:
I'll give you the broken OC kernel, so you can have tons of gigabytes instead of 2 :D:D

koquantam, 2016-04-09:
Do you have any ideas about bind mounting with Android fstab?

koquantam, 2016-06-11:
Bro, any ideas about this?
http://hastebin.com/izujazakah.xml

koquantam, 2016-06-11:
Nevermind, I found the problem

ngoquang2708, 2016-06-11:
Is that ```gralloc```?

koquantam, 2016-06-11:
Nope
I was about making a flag in defconfig that only Marshmallow and Lollipop have to use Permissive SELinux, but fails. Now when I boot CM13, it was in "Enforcing", no wonder it can't boot :D

koquantam, 2016-06-11:
Now let's fix other bugs left. My exams are over

koquantam, 2016-06-11:
Now I got good news here: My soft keys also broken, because I get the phone to the swimming pool :D

You got a good friend, bro :D

koquantam, 2016-06-11:
Have to increase zRAM size to avoid memory leak, at least 256MB is enough

ngoquang2708, 2016-06-11:
LOL

ngoquang2708, 2016-06-11:
I have to remove all HAL modules except ```hwcomposer``` and ```gralloc``` in order to keep the phone in a stable state, not reboot everytime it run out of memory due to memory leak.

koquantam, 2016-06-11:
All of *.scx15.so?

koquantam, 2016-06-11:
You used SHIM right?

koquantam, 2016-06-11:
Have to say, after booted you must kill SystemUI before it grows memory
Else after 10-20mins it'll kill itself, then the phone become normal

ngoquang2708, 2016-06-11:
I am investigating the cause of that memory leak.

koquantam, 2016-06-11:
You should see this: https://code.google.com/p/android/issues/detail?id=195104

ngoquang2708, 2016-06-11:
OK, our media codecs cause it.

koquantam, 2016-06-11:
Anything in logcat?
And I see VPX bug. Is that so?

ngoquang2708, 2016-06-11:
I think media server keep trying to load those media codecs and failed.

ngoquang2708, 2016-06-11:
Need some patch in source. But, make audio work first :)

ngoquang2708, 2016-06-11:
I pushed all of it.

ngoquang2708, 2016-06-11:
What funny is that the Cyanogenmod File Manger app does not see the SDCard, but Setting's Storage & USB see :)

koquantam, 2016-06-11:
You missed this flag, eh?
encryptable=userdata
That's why CM FM can't detect storages but Settings does. Now it does :D

koquantam, 2016-06-11:
Now it can't remount when unmounted

ngoquang2708, 2016-06-11:
OK, confirmed that CM FM see it. And Settings CAN remount it :), check again to see if there is any differences.

koquantam, 2016-06-11:
Wrong sysfs path :D
Applied your commit

koquantam, 2016-06-11:
Marshmallow is much more smooth than Lollipop, at least with me

ngoquang2708, 2016-06-11:
I though I the only one feel it :)

koquantam, 2016-06-11:
Bro, can you add prebuilt chromium commit?

ngoquang2708, 2016-06-11:
There is no chromium source, they already built it in external/chromium-webview. Read README.

koquantam, 2016-06-11:
Bro, have a look at CM13 dmesg, do grep with "ril", you'll see bunch of error messages there

ngoquang2708, 2016-06-11:
Yes. See https://github.com/ngoquang2708/android_device_samsung_vivalto3gvn/issues/42.

[81 more replies]

closed 2016-08-01 by ngoquang2708