ngoquang2708/android_device_samsung_vivalto3gvn #31: cm-12.1: "SIM cards changed" at every reboot

ngoquang2708, 2016-03-18:


koquantam, 2016-03-22:
You can close this, that doesn't exist
Btw, RIL signals seems buggy. We have internet access directly, but it shows exclamation marks

ngoquang2708, 2016-03-22:
I don't know, I had it before, now it is gone!!??
About RIL signal, You got it when you turn mobile data on or off?

koquantam, 2016-03-22:
After turning off WiFi, it acts like that

koquantam, 2016-03-22:
But still have internet access
Another mention: Only like that when wifi's connected, else doesn't

ngoquang2708, 2016-03-22:
It seems it is related to service provider.
I have another problem, when making a call, one sim doesn't have sound, I've to hang up and try again, the other sim doesnt have that. When swap sim slots, still have problem with that sim. So service provider case troubles here.

koquantam, 2016-03-22:
Maybe

koquantam, 2016-03-22:
Well, that's my fault after sharing the ROM on tinhte. Now Vietnamese everywhere :D

ngoquang2708, 2016-03-22:
:D

ngoquang2708, 2016-03-26:
Still persist, but randomly though.

koquantam, 2016-03-26:
Check your SIM card pins
Maybe they're dirty

ngoquang2708, 2016-03-26:
Haha, that a trick.

ngoquang2708, 2016-07-22:
CM12.1 is EOF.

closed 2016-07-22 by ngoquang2708