ngoquang2708/android_device_samsung_vivalto3gvn #32: cm-11.0: cannot install any font

ngoquang2708, 2016-03-26:
The installation is success. But when cannot switch to new font.
Logcat:
```
I/Timeline( 6522): Timeline: Activity_launch_request id:com.android.settings time:2508306
I/ActivityManager(  545): START u0 {cmp=com.android.settings/.Display} from pid 6522
D/gralloc.scx15(  126): alloc_device_alloc w:480, h:800, format:1 usage:0xf02 start
D/gralloc.scx15(  126): the flag 0x4 and the vadress is 0xb4e7e000 and the size is 0x177000
D/gralloc.scx15(  126): alloc_device_alloc handle:0xb8580db0 end err is 0
D/AndroidRuntime( 6348): Shutting down VM
W/dalvikvm( 6348): threadid=1: thread exiting with uncaught exception (group=0x41619ce0)
E/AndroidRuntime( 6348): FATAL EXCEPTION: main
E/AndroidRuntime( 6348): Process: com.android.settings, PID: 6348
E/AndroidRuntime( 6348): java.lang.RuntimeException: Unable to instantiate activity ComponentInfo{com.android.settings/com.android.settings.Display}: java.lang.ClassNotFoundException: Didn't find class "com.android.settings.Display" on path: DexPathList[[zip file "/system/framework/org.cyanogenmod.hardware.jar", zip file "/system/priv-app/Settings.apk"],nativeLibraryDirectories=[/vendor/lib, /system/lib]]
E/AndroidRuntime( 6348): 	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2131)
E/AndroidRuntime( 6348): 	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2271)
E/AndroidRuntime( 6348): 	at android.app.ActivityThread.access$800(ActivityThread.java:144)
E/AndroidRuntime( 6348): 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1205)
E/AndroidRuntime( 6348): 	at android.os.Handler.dispatchMessage(Handler.java:102)
E/AndroidRuntime( 6348): 	at android.os.Looper.loop(Looper.java:136)
E/AndroidRuntime( 6348): 	at android.app.ActivityThread.main(ActivityThread.java:5146)
E/AndroidRuntime( 6348): 	at java.lang.reflect.Method.invokeNative(Native Method)
E/AndroidRuntime( 6348): 	at java.lang.reflect.Method.invoke(Method.java:515)
E/AndroidRuntime( 6348): 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:732)
E/AndroidRuntime( 6348): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:566)
E/AndroidRuntime( 6348): 	at dalvik.system.NativeStart.main(Native Method)
E/AndroidRuntime( 6348): Caused by: java.lang.ClassNotFoundException: Didn't find class "com.android.settings.Display" on path: DexPathList[[zip file "/system/framework/org.cyanogenmod.hardware.jar", zip file "/system/priv-app/Settings.apk"],nativeLibraryDirectories=[/vendor/lib, /system/lib]]
E/AndroidRuntime( 6348): 	at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:56)
E/AndroidRuntime( 6348): 	at java.lang.ClassLoader.loadClass(ClassLoader.java:497)
E/AndroidRuntime( 6348): 	at java.lang.ClassLoader.loadClass(ClassLoader.java:457)
E/AndroidRuntime( 6348): 	at android.app.Instrumentation.newActivity(Instrumentation.java:1061)
E/AndroidRuntime( 6348): 	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2122)
E/AndroidRuntime( 6348): 	... 11 more
W/ActivityManager(  545): Process com.android.settings has crashed too many times: killing!
W/ActivityManager(  545):   Force finishing activity com.android.settings/.Display
I/ActivityManager(  545): Killing 6348:com.android.settings/1000 (adj 0): crash
```

thanhthuy89, 2016-03-27:
waiting

thanhthuy89, 2016-03-27:
cm12.1 bootloop when select font

koquantam, 2016-03-29:
They're installable, just after reboots, they're all acting dumb, just like VNTime with Unicode :D

ngoquang2708, 2016-07-22:
EOL.

closed 2016-07-22 by ngoquang2708