ngoquang2708/android_device_samsung_vivalto3gvn #38: cm-13.0: fstab rework

koquantam, 2016-04-19:
Yes, since changes in Marshmallow, so storage depends on fstab a lot
storage_list.xml now useless (removed from framework code)
Also, external fstab need this flag to be used by adoptable storage: encryptable=userdata
See here: https://source.android.com/devices/storage/config.html

ngoquang2708, 2016-04-19:
Updated c7c99c2c019a700637560177122064951685a9a1.
About adoptable storage, I will do it later when the device successfully booted. Currently, it cannot boot at all, just reboot to recovery.

koquantam, 2016-04-19:
SELinux also needs to be updated, in kernel implementations, because it's now in Android mainline

koquantam, 2016-04-19:
Can you give me the ROM zip?

ngoquang2708, 2016-04-19:
I think it is useless now. No Boot animation, no logcat, nothing.

koquantam, 2016-04-19:
Reboots straight to recovery?

ngoquang2708, 2016-04-19:
Yes

ngoquang2708, 2016-04-19:
CM 13.0 supports radio?!!

ngoquang2708, 2016-04-19:
Any idea on how to get system logs (dmesg, logcat) when the device auto reboot to recovery mode?

koquantam, 2016-04-19:
Nope, you can't
At least it must shows the boot animation OR screen lowers the light (decompressing kernel completed) then you can get dmesg or logcat

koquantam, 2016-04-19:
Umm... radio? Which app?

ngoquang2708, 2016-04-19:
It is good if I have UART, even I don't know what is is :D

ngoquang2708, 2016-04-19:
I can see it compiles some codes have ```radio``` word.

koquantam, 2016-04-19:
You just need a cable, because you've a Raspberry Pi right? It can do UART jobs

ngoquang2708, 2016-04-19:
Doesn't have any experience with this UART thing.

koquantam, 2016-04-19:
I guess you shouldn't touch this if you don't want everything end up in pain :D

ngoquang2708, 2016-04-19:
OK

ngoquang2708, 2016-04-19:
It just stop here: https://github.com/ngoquang2708/android_system_core/blob/cm-13.0/init/init.cpp#L988, cannot go further :(

ngoquang2708, 2016-04-19:
OK, It seems that the kernel do not know the format of ```sepolicy``` file that is generated by CM-13.0. I use the one in ```rootdir``` and the device now stuck at SS logo, not auto reboot any more.

koquantam, 2016-04-20:
There still lots of if conditions that triggers security_failure() command, including restorecon("/init"), write("...checkreqprot"...)
https://github.com/ngoquang2708/android_system_core/blob/cm-13.0/init/init.cpp#L994
Let's see if this kernel patch work for you:
https://github.com/SandPox/android_kernel_samsung_kyleproxx/commit/0fe93a236f1d35875310b6bde82621ac58ae298e

Qiangong2, 2016-04-22:
I'm having the same issue with my cm13 build for samsung galaxy star pro S7262 (logan2g)

koquantam, 2016-04-29:
@ngoquang2708 Bro, I think we should use CM12.1's init binary instead

koquantam, 2016-04-29:
CM13's init binary got fvcked up

Qiangong2, 2016-04-29:
@koquantam the patch you linked to before doesn't work

koquantam, 2016-04-29:
@Qiangong2 Can you be able to get logcat?

Qiangong2, 2016-04-29:
@koquantam it doesn't even build

ngoquang2708, 2016-05-04:
Anyone know how to use UART. I planned to buy a UART to USB convert to debug the kernel. It is CP2012. It should be Bus Pirate but the CP2012 is just about $3, Bus Pirate is $40 and I cannot effort. Then we need to probe the UART location in the phone main board. But one thing make I afraid is that what is the console name the phone use to direct its output to. proberly ```/dev/ttyS1``` but while the phone is running, ```/proc/cmdline``` doesn't have it.

koquantam, 2016-05-07:
Well, I don't think our country has those UART thingy, because it's not popular here
Anyway, how do you build kernel first in CM source?

ngoquang2708, 2016-05-08:
@koquantam Are you kidding man, we have tons of it [here](http://hshop.vn/products/mach-chuyen-usb-uart-cp2102).
To build the kernel first, run ```make bootimage```. Do not use any commands that use parallel building like ```make -jX``` or ```mka```.

ngoquang2708, 2016-05-08:
BONUS: I broke my softkeys while opening the screen, such a genius.

koquantam, 2016-05-08:
LOL, the whole mainboard is glued, not screws

koquantam, 2016-05-08:
Just reconnect the connection cable

ngoquang2708, 2016-05-08:
Did it many times but didn't help, using your ```qemu.hw.mainkeys=0``` trick.

ngoquang2708, 2016-05-09:
Cannot locate the UART location on the board :(

ngoquang2708, 2016-05-24:
@Qiangong2 Maybe one of these commits will help you: 6d253558fe441b4886e503e77ac53f83313998bd and bbc01d8

ngoquang2708, 2016-05-24:
Commit bbc01d8

closed 2016-05-24 by ngoquang2708