ngoquang2708/android_device_samsung_vivalto3gvn #39: cm-12.1: Random reboots when using mobile data for too long

koquantam, 2016-05-21:
Again, users complaints

Off-topic: Did you retired bro?

ngoquang2708, 2016-05-22:
Maybe the phone is overheated because of OC kernel.

ngoquang2708, 2016-05-22:
Good news
[image]
[image]

But crash every where, you known :)

koquantam, 2016-05-22:
Wow, good work bro :D

koquantam, 2016-05-22:
But, what about the soft keys?

ngoquang2708, 2016-05-23:
Don't ask about the soft keys! Are you kidding me?!!

ngoquang2708, 2016-05-23:
:D

koquantam, 2016-05-24:
Vibrator died :D

ngoquang2708, 2016-05-24:
What?

koquantam, 2016-05-24:
My phone's vibrator has died

ngoquang2708, 2016-05-24:
Did you kill it :D

koquantam, 2016-05-24:
Nope :D
My brother plays games with the phone, and he got it pulled out from his wet hand while running, so.....

koquantam, 2016-05-24:
Ok, how was the progress with CM13?
Any important news?

ngoquang2708, 2016-05-24:
Trying to fix the radio first.

koquantam, 2016-05-28:
Bro, IDK why people complaints a lot about Beta 3
Even installed beta 2 boot.img they still having random reboots, espically when they using mobile data
Can you remake it? :D
Beta 3 just have dlmalloc() impl for ART, updated sources (no need), some WITH_DEXPREOPT flags and removed "--block" parameters from build/core/Makefile for easy modding

Maybe my /home partition is corrupted

ngoquang2708, 2016-05-28:
Ok, I will build new one

koquantam, 2016-05-28:
Wait, classes.dex still being relocated to /data/dalvik-cache

koquantam, 2016-05-28:
Bro, you missed this flag
`WITH_DEXPREOPT_COMP := false`

closed 2016-06-11 by koquantam