ngoquang2708/android_device_samsung_vivalto3gvn #41: cm-13.0: audio not work

ngoquang2708, 2016-06-11:
Audio is not work because the audio module cannot access ```/dev/snd/controlC%d``` due to permission denies. Need to file out what service load audio module and set a proper user or group for it. My first guess is ```mediaserver```, but it already had ```system``` group, which is ```/dev/snd/controlC%d``` has. Don't know why audio module cannot access it though.

ngoquang2708, 2016-06-11:
Working now with this 71d544bfe690c8cec1699e4988bea30358a0b28a. Need a better solution.

ngoquang2708, 2016-06-11:
Fixed by https://github.com/ngoquang2708/android_system_core/commit/838a95a878be4b1f708fff64a7edba3c57e09057.

ngoquang2708, 2016-06-13:
Stop working again after rebase upstream.

ngoquang2708, 2016-06-13:
@koquantam This thing look like crap again :(

koquantam, 2016-06-13:
We're back with CM-11.0 :D

koquantam, 2016-06-14:
Did you tried REmarking mediaserver as root user?

ngoquang2708, 2016-06-14:
OMG, sound back again. I hate it so much.

koquantam, 2016-06-14:
How? Randomly?

ngoquang2708, 2016-06-14:
Recompile the source again, and it work again. Don't know why :(

ngoquang2708, 2016-06-23:
Found some interesting things. We didn't use ```audio_policy.*.so``` at all since CM-12.1. Deleted all of them, and sound stills working.
Then, I discovered that, CM-12.1 use ```libaudiopolicymanagerdefault.so```. So we should focus on them.
The source is located at ```frameworks/av/services/audiopolicy```.

koquantam, 2016-06-24:
Lol, I known that lib :D but thought it was in hardware frameworks

ngoquang2708, 2016-06-26:
Found some thing interesting. I can route audio from mic to speaker by echo 1,2,7 > /dev/pipe/mmi.audio.ctrl, route to earpiece by echo 1,1,7 > /dev/pipe/mmi.audio.ctrl. I don't known the way android route app's sound to speaker. Any ideas?
@koquantam 
@corphish

ngoquang2708, 2016-06-27:
Seem this fix our issue completely https://github.com/ngoquang2708/android_device_samsung_vivalto3gvn/commit/0f9e32856267bac253ccb3bd07dc869604177464.
I try 1 clean build with that flag enabled and another clean build with that flag disabled to confirm it.
@koquantam If you have time, pleases confirm this.

koquantam, 2016-06-27:
OK, CM12.1 is done with me, now I'm doing for our vivalto

koquantam, 2016-06-27:
Making clean build now

koquantam, 2016-06-27:
This situation was just like RIL :D
One BoardConfig flag solves all

ngoquang2708, 2016-06-27:
Hope so :)

koquantam, 2016-06-27:
Confirmed

koquantam, 2016-06-27:
I think we already know this happen while reading the commit message from your reference commit

koquantam, 2016-06-27:
Hmm... somehow audio doesn't work again o.O

ngoquang2708, 2016-06-28:
Aggh, just happened to me with the last build, again. Doing a clean build, again :(

ngoquang2708, 2016-06-28:
Since CM-11.0, this audio issue is always suck.

koquantam, 2016-06-28:
Totally sucks

ngoquang2708, 2016-06-28:
But at least it works for you the first time, right :)

koquantam, 2016-06-28:
I guess, this is our problem
`06-28 11:50:49.637   251   251 E TinyAlsaUtils: open snd control failed.`

ngoquang2708, 2016-06-28:
I just restart an audio is working again, what the heck!

koquantam, 2016-06-28:
Oh you've told in description :D

ngoquang2708, 2016-06-28:
About this:
```
06-28 12:01:30.284   148   148 E TinyAlsaUtils: open snd control failed: No such file or directory
```
I have it too, it just cannot open one card, we have 3:
```
06-28 12:01:30.284   148   148 I audio_hw_primary: s_tinycard = 0, s_vaudio = -1,s_voip = 2, s_bt_sco = 3,s_vaudio_w is 1
```

koquantam, 2016-06-28:
Should we use again the mediaserver hack?

ngoquang2708, 2016-06-28:
```mediaserver``` already had the right permission to access ```/dev/snd/*```. The fact is that audio work for some times.
But we should try.
About my previous comment about sound working again after reboot, sorry, false alarm. I am using my other build that with audio working, not my latest build :(

ngoquang2708, 2016-06-28:
Can I ask you that your audio stop working with different builds or with the same build?

koquantam, 2016-06-28:
It work for the first time in the last build from yesterday, then stopped working after ONE RESTART LOL

ngoquang2708, 2016-06-28:
OK, one more clue.

koquantam, 2016-06-28:
That's all :D

koquantam, 2016-06-28:
If we have to touch mediaserver again, then I think it's init problem (system/core)

ngoquang2708, 2016-06-28:
Try this [magic.diff.zip](https://github.com/ngoquang2708/android_device_samsung_vivalto3gvn/files/336978/magic.diff.zip).

koquantam, 2016-06-28:
Is that real magic? :D

ngoquang2708, 2016-06-28:
Yes

koquantam, 2016-06-28:
OK, gonna try with my Core Prime :D

Seems everything in our tree works with it

ngoquang2708, 2016-06-28:
Do not need to do a clean build, just rebuild ```audio.primary.scx15.so```.

koquantam, 2016-06-28:
Right

koquantam, 2016-06-28:
Btw, how to rebuild a blob only?
Just "mka bacon" or something else?

ngoquang2708, 2016-06-28:
cd to that audio dir
then ```mm```

ngoquang2708, 2016-06-28:
Use ```mmp```, it will push to your phone automatically using ```adb```.

koquantam, 2016-06-28:
Wow, thanks

ngoquang2708, 2016-06-28:
You never read CM's Wiki, right :)

ngoquang2708, 2016-06-28:
Tons of great things there.

koquantam, 2016-06-28:
I just know basic parameters, like mka :D

ngoquang2708, 2016-06-28:
I've read it, and I missed that ```mka```, until you told me.

koquantam, 2016-06-28:
LOL

koquantam, 2016-06-28:
OK, confirmed this magic thingy :D

koquantam, 2016-06-28:
Now I can watch YouTube everyday :D

ngoquang2708, 2016-06-28:
Seem we lost the audio equalizer, make sound seem... I don't know which word to describe it, like cracked... :)

ngoquang2708, 2016-06-28:
I had that issue since CM-11.0, not happen in CM-12.1, then it come back in CM-13.0.

koquantam, 2016-06-28:
It seems OK for me, even watching videos

ngoquang2708, 2016-06-28:
Yes, listen to some music with loud bass, it will kill your speaker :)

koquantam, 2016-06-28:
Should we close this issue?

ngoquang2708, 2016-06-28:
No no... NOOO

ngoquang2708, 2016-06-28:
I am afraid of doing it :(

koquantam, 2016-06-28:
Ok, it's really our pain for long time, so nope :D

[12 more replies]

closed 2016-07-22 by ngoquang2708