ngoquang2708/android_device_samsung_vivalto3gvn #42: cm-13.0: ```rild``` crashes

ngoquang2708, 2016-06-11:
```
*** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
Build fingerprint: 'samsung/cm_vivalto3gvn/vivalto3gvn:6.0.1/MOB30D/edd9e1365f:userdebug/test-keys'
Revision: '0'
ABI: 'arm'
pid: 2853, tid: 2872, name: rild  >>> /system/bin/rild <<<
signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x8
    r0 00000042  r1 b60403e0  r2 73865c5e  r3 73865c5e
    r4 00000000  r5 00000b38  r6 00000000  r7 b7760304
    r8 b6849980  r9 b684978c  sl b5f42000  fp b6f3e769
    ip b6040408  sp b60408e0  lr b6f385a9  pc b6f385a8  cpsr 60000030
    d0  676f6c522d657352  d1  0000000000000049
    d2  000000000000004c  d3  0000000000000043
    d4  4352554f535f4e4f  d5  20736c6961662045
    d6  4441525f45207962  d7  415f544f4e5f4f49
    d8  0000000000000000  d9  0000000000000000
    d10 0000000000000000  d11 0000000000000000
    d12 0000000000000000  d13 0000000000000000
    d14 0000000000000000  d15 0000000000000000
    d16 0000000000000000  d17 0000000000000000
    d18 0000000000000000  d19 0000000000000000
    d20 0000000000000000  d21 0000000000000000
    d22 0000000000000000  d23 0000000000000000
    d24 0000000000000000  d25 0000000000000000
    d26 0000000000000000  d27 0000000000000000
    d28 0000000000000000  d29 0000000000000000
    d30 0000000000000000  d31 0000000000000000
    scr 00000000

backtrace:
    #00 pc 000055a8  /system/lib/libril.so (android::CommandThread(void*)+39)
    #01 pc 0000b7dd  /system/lib/libril.so (thread_pool_worker(void*)+116)
    #02 pc 000417fb  /system/lib/libc.so (__pthread_start(void*)+30)
    #03 pc 00019325  /system/lib/libc.so (__start_thread+6)

stack:
         b60408a0  00000000
         b60408a4  b77605d0  [heap]
         b60408a8  00000010
         b60408ac  00000010
         b60408b0  00000010
         b60408b4  00000000
         b60408b8  00000000
         b60408bc  73865c5e
         b60408c0  00000000
         b60408c4  00000000
         b60408c8  00000b38
         b60408cc  00000000
         b60408d0  b7760304  [heap]
         b60408d4  b6f385a9  /system/lib/libril.so (android::CommandThread(void*)+40)
         b60408d8  b6f3fb04  /system/lib/libril.so
         b60408dc  00000077
    #00  b60408e0  00000b38
         b60408e4  b6f38581  /system/lib/libril.so (android::CommandThread(void*))
         b60408e8  00000000
         b60408ec  b7760300  [heap]
         b60408f0  b7760528  [heap]
         b60408f4  b6f3e7df  /system/lib/libril.so (thread_pool_worker(void*)+118)
    #01  b60408f8  575c598d
         b60408fc  00000000
         b6040900  b5e3e000  [anon:thread signal stack]
         b6040904  b6040930  [stack:2872]
         b6040908  b6040970  [stack:2872]
         b604090c  b6040930  [stack:2872]
         b6040910  00000078
         b6040914  b6f087fd  /system/lib/libc.so (__pthread_start(void*)+32)
    #02  b6040918  b6f087dd  /system/lib/libc.so (__pthread_start(void*))
         b604091c  b6040938  [stack:2872]
         b6040920  b6f087dd  /system/lib/libc.so (__pthread_start(void*))
         b6040924  b6ee0327  /system/lib/libc.so (__start_thread+8)
[... 377 lines cut ...]
    b6f6f000-b6f6ffff rw-         0      1000  [anon:linker_alloc_32]
    b6f70000-b6f70fff r--         0      1000  [anon:linker_alloc]
    b6f71000-b6f90fff r--         0     20000  /dev/__properties__
    b6f91000-b6f91fff r--         0      1000
    b6f92000-b6f92fff ---         0      1000
    b6f93000-b6f94fff rw-         0      2000  [anon:thread signal stack]
    b6f95000-b6f95fff r-x         0      1000  [sigpage]
    b6f96000-b6fb2fff r-x         0     1d000  /system/bin/linker (BuildId: cff1f566d4617f529e8b9435a8740fc0)
    b6fb3000-b6fb3fff r--     1c000      1000  /system/bin/linker
    b6fb4000-b6fb5fff rw-     1d000      2000  /system/bin/linker
    b6fb6000-b6fb7fff rw-         0      2000
    b6fb8000-b6fb9fff r-x         0      2000  /system/bin/rild
    b6fba000-b6fbafff r--      1000      1000  /system/bin/rild
    b6fbb000-b6fbbfff rw-      2000      1000  /system/bin/rild
    b775a000-b7765fff rw-         0      c000  [heap]
    beac0000-beae0fff rw-         0     21000  [stack]
    ffff0000-ffff0fff r-x         0      1000  [vectors]
--------- tail end of log main
06-11 04:47:06.596  2853  2858 D         : set_multiclient_max_fd: maxfd=15
06-11 04:47:06.596  2853  2858 D         : set_multiclient_fd_set:
06-11 04:47:06.596  2853  2858 D         : set_multiclient_fd_set: fd=15
06-11 04:47:09.632  2853  2871 I libc    : Another thread contacted debuggerd first; not contacting debuggerd.
06-11 04:47:09.633  2853  2873 I libc    : Another thread contacted debuggerd first; not contacting debuggerd.
--- --- --- --- --- --- --- --- --- --- --- --- --- --- --- ---
```

ngoquang2708, 2016-06-11:
```rild``` would not crash if we disable all telephony service in android runtime (include ```full_base.mk``` instead of ```full_base_telephony.mk``` in ```device_vivalto3gvn.mk```).

koquantam, 2016-06-11:
Have a look at build/target/product/telephony.mk

ngoquang2708, 2016-06-11:
Yeah, I think there are some changes in M API crash our ```rild```.

koquantam, 2016-06-11:
Oops, I've deleted my comment :D

koquantam, 2016-06-11:
You've to look up the crash log, because crash message is there

koquantam, 2016-06-11:
Weird here: I can't see any rild crash logs in my logcat

ngoquang2708, 2016-06-11:
```/data/tombstones```

koquantam, 2016-06-15:
It's very weird that logcat doesn't give any signs of  error for rild, just say rild has crashed, with debug backtrace

ngoquang2708, 2016-06-15:
This is the hardest crash type to debug since it belongs to the logic of the blob. Just like a case in the past, where the naming conversion for system prop that the ```rild``` use is difference from the one that the framework use.

koquantam, 2016-06-16:
Bro, I think we get the problem now
Our rild use RIL_register instead of RIL_register_socket. Native AOSP rild doesn't have that too, so maybe missing that symbol make rild crashes, I think
This picture down here is when executing AOSP rild
[image]

koquantam, 2016-06-16:
This `page record for 0xb6f6a00c was not found (block_size=64)` exist a lot in our logcat

koquantam, 2016-06-17:
Fixed as this: https://github.com/koquantam/android_device_samsung_vivalto3gvn/commit/e5cfc21660dfab84dc05e12553e2acfddda2da56

koquantam, 2016-06-17:
Look here, baseband and signals are found!
[image]

koquantam, 2016-06-17:
Fun fact: Audio is not working, but we can hear people saying while calling :D
Oh well, it's audio RIL, so it works, right?

ngoquang2708, 2016-06-17:
CM12.1 EOL is counting.

closed 2016-06-17 by ngoquang2708

koquantam, 2016-06-17:
Bro, we have another problem
After reboot, SystemUI and com.android.phone crashed :D

ngoquang2708, 2016-06-17:
That is the work for tomorrow.I am sleeping. So you do.

koquantam, 2016-06-17:
OK, but I found the error. It's binder again. In logcat:
`06-18 01:30:31.259 20926 20941 E JavaBinder: !!! FAILED BINDER TRANSACTION !!!  (parcel size = 1744)`

ngoquang2708, 2016-06-18:
You should open new issue about crashing of Phone and SystemUI.