ngoquang2708/android_device_samsung_vivalto3gvn #46: cm-13.0: Screen flickering

koquantam, 2016-06-14:
Sometimes the phone does that, but with the affected apps only, even with disabled "Strict mode". Quick solution is tapping on recents button, then cone back with the current opening app

ngoquang2708, 2016-06-15:
Confirmed.

koquantam, 2016-06-28:
Now is this most annoying issue

koquantam, 2016-06-28:
I guess CM13 developer options are broken

ngoquang2708, 2016-06-28:
Or our HALs ar broken :)

koquantam, 2016-06-29:
I guess again, gralloc's fault

ngoquang2708, 2016-07-06:
May be this commit will fix this issue https://github.com/ngoquang2708/android_device_samsung_vivalto3gvn/commit/5b47381b39bf9be47ede313f7539004c5028b272.

koquantam, 2016-07-06:
So, do we need to revert this? https://github.com/koquantam/android_frameworks_base/commit/1003c499017267bc497fc45cf8802cb933d9781b

ngoquang2708, 2016-07-06:
This Gerrit review http://review.cyanogenmod.org/#/c/125226/ is abandoned so we should not revert your mentioned commit. I just gave a ref from the i9300 device which has the same issue as us.

ngoquang2708, 2016-07-07:
I hasn't had any flickering since that change so I will close this issue.

closed 2016-07-07 by ngoquang2708

koquantam, 2016-07-07:
Oh well, I would test so

koquantam, 2016-07-07:
But if I had, you have to open it again :D

ngoquang2708, 2016-07-07:
OK

koquantam, 2016-07-07:
Anyway, do you plan making Nougat for our phone?

ngoquang2708, 2016-07-07:
I will wait until we have its source codes :)

koquantam, 2016-07-07:
Yes, I'll wait too :D
Our country needs people like you

koquantam, 2016-07-07:
Confirmed

koquantam, 2016-07-07:
It's very nice that we don't face black preview recents like CM12.1 in CM13 anymore

ngoquang2708, 2016-07-07:
Codecs work great in CM13 with minor modification :)

koquantam, 2016-07-07:
Actually, stock gralloc also does that change :v

ngoquang2708, 2016-07-07:
You mean the fix for ScreenRecording? I haven't found the root cause of
that issue yet.

koquantam, 2016-07-07:
Yes, screen recording

ngoquang2708, 2016-07-07:
But that was fixed
https://github.com/ngoquang2708/android_device_samsung_vivalto3gvn/commit/67ad84476b4044701a869c51582ae744bdd0fab2

koquantam, 2016-07-07:
These HALs you took from `sp7715ga_gonk4.4_hemiao_mp` branch right?
So this is why it doesn't work with me

ngoquang2708, 2016-07-07:
Yes. Your board is sc8830, right?

koquantam, 2016-07-07:
Yes. The `sprdroid4.4` branch works for me, except `gralloc`
It gives me an error, like `deleted function` which I never faced before

ngoquang2708, 2016-07-07:
Give me the compile error log, I could help if I can.

koquantam, 2016-07-07:
Here
[image]

ngoquang2708, 2016-07-07:
I know that bug, compare my gralloc framebuffer_device.cpp, you will see
the differences.

koquantam, 2016-07-07:
Thanks bro. Now its usable now

koquantam, 2016-07-07:
Bro, I think we should switch all libs changes to vendor tree
Now we don't have same devices anymore, but at least our devices has same SoC types that we can easily sync our changes on

ngoquang2708, 2016-07-07:
OK, I will plan it.

ngoquang2708, 2016-07-07:
But don't use my ```android_vendor_sprd_open-source``` tree. I just use it to view the history.
We should make ```android_hardware_samsung_sprd-common``` tree instead.

koquantam, 2016-07-07:
Oh, you're right
But I've just pushed a commit on vendor tree for an hour :D

ngoquang2708, 2016-07-07:
I planed to fork @TeamButter sprd-common tree, but there are too many conflict so I afraid of doing that :)

koquantam, 2016-07-07:
Well most of their tree are for sc8810, while ours are scx15 and sc8830
But anyway, can we make separated folder for each SoC?

ngoquang2708, 2016-07-07:
Yes. And another tree ```android_device_samsung_sprd-common```.

koquantam, 2016-07-07:
Bro, I remember you've fixed wrong color with camera using H264 AVC codec
Can you locate me to that commit?

ngoquang2708, 2016-07-07:
https://github.com/ngoquang2708/android_device_samsung_vivalto3gvn/commit/245e30afc7fa164b0d1e356c8529c8f8d694b234

koquantam, 2016-07-09:
I've made a common hardware tree. Can you check it out and use it?