ngoquang2708/android_device_samsung_vivalto3gvn #47: cm-13.0: SystemUI, com.android.phone and Messaging crash

koquantam, 2016-06-18:


ngoquang2708, 2016-06-18:
Crash here:
```
06-18 17:23:42.412   682  1187 E AndroidRuntime: FATAL EXCEPTION: PhoneStatusBar
06-18 17:23:42.412   682  1187 E AndroidRuntime: Process: com.android.systemui, PID: 682
06-18 17:23:42.412   682  1187 E AndroidRuntime: Theme: themes:{}
06-18 17:23:42.412   682  1187 E AndroidRuntime: java.lang.IllegalArgumentException: key.length > 31
06-18 17:23:42.412   682  1187 E AndroidRuntime: 	at android.os.Parcel.readException(Parcel.java:1624)
06-18 17:23:42.412   682  1187 E AndroidRuntime: 	at android.os.Parcel.readException(Parcel.java:1573)
06-18 17:23:42.412   682  1187 E AndroidRuntime: 	at com.android.internal.telephony.ISub$Stub$Proxy.getActiveSubscriptionInfoList(ISub.java:604)
06-18 17:23:42.412   682  1187 E AndroidRuntime: 	at android.telephony.SubscriptionManager.getActiveSubscriptionInfoList(SubscriptionManager.java:632)
06-18 17:23:42.412   682  1187 E AndroidRuntime: 	at com.android.systemui.statusbar.policy.NetworkControllerImpl.doUpdateMobileControllers(NetworkControllerImpl.java:435)
06-18 17:23:42.412   682  1187 E AndroidRuntime: 	at com.android.systemui.statusbar.policy.NetworkControllerImpl.updateMobileControllers(NetworkControllerImpl.java:430)
06-18 17:23:42.412   682  1187 E AndroidRuntime: 	at com.android.systemui.statusbar.policy.NetworkControllerImpl.registerListeners(NetworkControllerImpl.java:217)
06-18 17:23:42.412   682  1187 E AndroidRuntime: 	at com.android.systemui.statusbar.policy.NetworkControllerImpl.-wrap0(NetworkControllerImpl.java)
06-18 17:23:42.412   682  1187 E AndroidRuntime: 	at com.android.systemui.statusbar.policy.NetworkControllerImpl$1.run(NetworkControllerImpl.java:810)
06-18 17:23:42.412   682  1187 E AndroidRuntime: 	at android.os.Handler.handleCallback(Handler.java:739)
06-18 17:23:42.412   682  1187 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:95)
06-18 17:23:42.412   682  1187 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:148)
06-18 17:23:42.412   682  1187 E AndroidRuntime: 	at android.os.HandlerThread.run(HandlerThread.java:61)
```

ngoquang2708, 2016-06-18:
Look like the cause is the limitation of android system prop which has max length for key of 31.

ngoquang2708, 2016-06-18:
This commit cause something trouble https://github.com/ngoquang2708/android_frameworks_base/commit/21301f6016ce50594f6cd9c66581ee9a08b57be9.

koquantam, 2016-06-18:
You mean the function syntax is too long?

koquantam, 2016-06-18:
But... why the system boots well on first boot, then starting broken on next boots? Even a soft reboot

koquantam, 2016-06-18:
And it throws a failed binder transaction error

koquantam, 2016-06-18:
Can you have a look at this commit?
https://github.com/CyanogenMod/android_frameworks_base/commit/0450565d9a6de0f0828c9ad8ce16c8645c6f394c

koquantam, 2016-06-19:
@corphish
Can you help us at this?

ngoquang2708, 2016-06-19:
I observed these things:
- When the rild successfully load, it sends some thing the the android runtime
- The android runtime just store those information at that time so it will not crash at that time
- On reboot, the android read back those information and crash even if I remove the rild binary
- Factory reset solves this for first boot only

koquantam, 2016-06-19:
I didn't observed the second thing

ngoquang2708, 2016-06-19:
Maybe this should fix this issue https://github.com/ngoquang2708/android_frameworks_base/commit/be8378d1b54bf4c68af5313b7f21826d3859efc4. Compiling.

ngoquang2708, 2016-06-19:
My mistake.

koquantam, 2016-06-19:
Looks like it just a checking procedure, nothing else

ngoquang2708, 2016-06-19:
Just a the first glance.

koquantam, 2016-06-19:
Fixed as this: https://github.com/koquantam/android_frameworks_base/commit/f40670ca08551e87660eb984fb52a9c17bb98f9f

closed 2016-06-19 by koquantam

ngoquang2708, 2016-06-20:
This https://github.com/ngoquang2708/android_frameworks_base/commit/da2fe5715a4c96239f0365ca0b0714dd08cb57e8 and this https://github.com/ngoquang2708/android_frameworks_base/commit/9aeba709fb1148adc4bf20542bbc812c12a8d754 should solve this issue. No more hacks.