ngoquang2708/android_device_samsung_vivalto3gvn #48: cm-13.0: WiFi auto-disconnect

koquantam, 2016-06-20:
Sometimes this happens, mostly when the phone is at sleep state, even "Always turn on WiFi when sleep" is "always" or "Charging only"

ngoquang2708, 2016-07-30:
Logs from @XDA/adriansticoid:

- [event.txt](https://github.com/ngoquang2708/android_device_samsung_vivalto3gvn/files/391842/event.txt)
- [logcat.txt](https://github.com/ngoquang2708/android_device_samsung_vivalto3gvn/files/391843/logcat.txt)

ngoquang2708, 2016-07-30:
Seem ```IpReachabilityMonitor``` auto disconnect after failed some attempts to ping the gateway.
My logcat:
```
07-30 18:03:43.037  2278  2278 I wpa_supplicant: wlan0: WPA: Group rekeying completed with 14:cc:20:14:e9:ef [GTK=TKIP]
07-30 18:03:43.042   467   625 D IpReachabilityMonitor: probing ip=192.168.252.1%8
07-30 18:03:43.051   467  2286 D NetlinkSocketObserver: NeighborEvent{elapsedMs=5088471, 192.168.252.1, [(null)], RTM_NEWNEIGH, NUD_FAILED}
07-30 18:03:43.051   467  2286 W NetlinkSocketObserver: ALERT: NeighborEvent{elapsedMs=5088471, 192.168.252.1, [(null)], RTM_NEWNEIGH, NUD_FAILED}
07-30 18:03:43.073  2278  2278 I wpa_supplicant: wlan0: WPA: Group rekeying completed with 14:cc:20:14:e9:ef [GTK=TKIP]
07-30 18:03:43.090   467   625 D IpReachabilityMonitor: probing ip=192.168.252.1%8
07-30 18:03:43.092   467  2286 W IpReachabilityMonitor: FAILURE: LOST_PROVISIONING, NeighborEvent{elapsedMs=5088471, 192.168.252.1, [(null)], RTM_NEWNEIGH, NUD_FAILED}
07-30 18:03:43.094   467   625 E IpReachabilityMonitor: Error probing ip=192.168.252.1%8, errmsg=NetlinkErrorMessage{ nlmsghdr{StructNlMsgHdr{ nlmsg_len{56}, nlmsg_type{2(NLMSG_ERROR)}, nlmsg_flags{0())}, nlmsg_seq{1}, nlmsg_pid{-4162} }}, nlmsgerr{StructNlMsgErr{ error{-22}, msg{StructNlMsgHdr{ nlmsg_len{36}, nlmsg_type{28(RTM_NEWNEIGH)}, nlmsg_flags{261(NLM_F_REQUEST|NLM_F_ACK|NLM_F_ROOT))}, nlmsg_seq{1}, nlmsg_pid{0} }} }} }
07-30 18:03:43.097   467   625 D IpReachabilityMonitor: clear: iface{wlan0/8}, v{5}, ntable=[]
07-30 18:03:43.108  2278  2278 I wpa_supplicant: wlan0: CTRL-EVENT-DISCONNECTED bssid=14:cc:20:14:e9:ef reason=3 locally_generated=1
07-30 18:03:43.118   467   625 D WifiStateMachine: WifiStateMachine: Leaving Connected state
07-30 18:03:43.119   467   625 D IpReachabilityMonitor: clear: iface{wlan0/8}, v{6}, ntable=[]
07-30 18:03:43.119   467   625 D WifiNative-HAL: stopRssiMonitoring, cmdId 0
```
REF: https://code.google.com/p/android/issues/detail?id=196035

ngoquang2708, 2016-07-30:
Fixed by https://github.com/ngoquang2708/android_device_samsung_vivalto3gvn/commit/773986dabc49a8cb550be657f9f78f61994109f8.

closed 2016-07-30 by ngoquang2708