ngoquang2708/android_device_samsung_vivalto3gvn #51: common: STK doesn't work

koquantam, 2016-07-13:
Tested on both core33g and vivalto. If you choose any option from the menu (except choosing SIM), STK stuck and continuously spinning

ngoquang2708, 2016-07-13:
Tested with single SIM. Not tested with dual SIM yet.

https://youtu.be/YBuqXccwNlo

PhantomAssassin12, 2016-07-13:
Confirmed

ngoquang2708, 2016-07-13:
What do you mean? Are you affected?

PhantomAssassin12, 2016-07-14:
Yeah tested on dual sim card

koquantam, 2016-07-14:
Hmm... I got some results after testing with different operators

The STK actually work for some network operators, in our country are Vinaphone and Mobifone, but not Viettel. Maybe VT has different way of sending STK signals. I'll check logcat with this

ngoquang2708, 2016-07-27:
I tested it on dual-sim. No issue.
@koquantam @PhantomAssassin12 Anyone stills affect by this issue? Should I close this?

koquantam, 2016-07-27:
Ok, pls close this

closed 2016-07-27 by koquantam