ngoquang2708/android_device_samsung_vivalto3gvn #53: Cannot use Location in 'Device only' mode

ngoquang2708, 2016-07-21:
This is a common CM13 issue with permission.
```
com.google.android.finsky.utils.aa.a(57): No permission to get location. java.lang.SecurityException: "gps" location provider requires ACCESS_FINE_LOCATION permission
```

ngoquang2708, 2016-07-23:
That error is gone when I grant read location permission for ```Google Play Store``` in ```Apps``` settings.

ngoquang2708, 2016-07-23:
@koquantam Do you have this issue?

koquantam, 2016-07-23:
No, I don't

PhantomAssassin12, 2016-07-23:
I don't have this issue my gps is working fine in device only.

koquantam, 2016-07-30:
@ngoquang2708 
Off topic: Can you check the battery section in Settings
For me I only see "Screen" there

ngoquang2708, 2016-07-30:
Where is your power_profile.xml come from? I extract it from my stock's frameworks-res.apk.

koquantam, 2016-07-30:
Yes also from my stock framework-res.apk

koquantam, 2016-07-30:
In CM12.1, battery stats working fine, but CM13 doesn't

ngoquang2708, 2016-11-19:
@PhantomAssassin12 If you have time, can you test GPS (device only mode since reboot) in CM14.1 and report me here. Thank you.