ngoquang2708/android_device_samsung_vivalto3gvn #54: WiFi-Direct does not work

ngoquang2708, 2016-07-21:


koquantam, 2016-07-22:
It must be "ifconfig" instead of ipconfig. It's for Windows only
And that p2p0 device is for USB tethering. Don't you see "Ethernet" there?
[image]

koquantam, 2016-07-22:
Weird is we get "No such file or directory" here

ngoquang2708, 2016-07-22:
OK, I misspelled it. Did you compare it with wlan0 (when wifi is off)? They
are the same!

ngoquang2708, 2016-07-22:
I ran ```wpa_supplicant -i p2p0 -Dnl80211 -c /data/misc/wifi/p2p_supplicant.conf``` to test it. With ```-i wlan0```, it works, with ```-i p2p0```, it does not.

koquantam, 2016-07-22:
OK, my mistake :D

koquantam, 2016-07-22:
Look at our init.wifi.rc, p2p_supplicant service use wlan0 interface lol

ngoquang2708, 2016-07-28:
WiFi-Direct is working now. Maybe CM's repo is updated with changes that make it works.

ngoquang2708, 2016-07-28:
[image]

closed 2016-07-28 by ngoquang2708