ngoquang2708/android_device_samsung_vivalto3gvn #56: Cannot get pass Setup Wizard on clean install

ngoquang2708, 2016-07-25:
On clean install, we cannot click any button.

ngoquang2708, 2016-07-25:
The current workaround is install older build, set it up then flash the latest build without factory reset.

thanhthuy89, 2016-07-26:
bác tạo cho e cái patch cho beta1update 1 ko bị screen flickering dc ko?. tuy am thanh rè nhung co the fix bang audio fx dc.am thanh to hon bản beta 2 nhieu tieng anh yeu nen cho e ghi tam it tieng viet

PhantomAssassin12, 2016-07-26:
Yeah its like locking the setup and the status pulled down and i can go settings without setting it up. Btw not whole screen but haft of it down side.

ngoquang2708, 2016-07-27:
@thanhthuy89 Gải nén file này rồi copy vô ```/system/lib/```: [libhwui.so.zip](https://github.com/ngoquang2708/android_device_samsung_vivalto3gvn/files/386179/libhwui.so.zip)

ngoquang2708, 2016-07-27:
@thanhthuy89 Âm thanh to qúa nên nó bị rè là phải.

thanhthuy89, 2016-07-27:
Lúc cài beta2 xong thay lỗi quay lai update1 flash xong quen gapp khoi dong lai vao twrp wipe data/factory flash gapp van ko bi đơ thanh trạng thái,sau lan nay lam vay kéo thang trai thái xuong nhung ko kéqo bật tắt nhanh dc la sau vay bac Quang

ngoquang2708, 2016-07-27:
@thanhthuy89 Bạn có backup beta1update1 ko, restore nó lại đi. Bạn không flash Gapps sau khi flash rom nên bị lỗi gì rồi.

thanhthuy89, 2016-07-27:
Noi thiet nhe minh flash beta1 up1 ca may chuc lan xong deu wipe data nhubg ko bi vay? Sau gio flash xong wipe data cai le keo xuong ko dc

ngoquang2708, 2016-07-27:
@thanhthuy89 Neu co y dinh cai lai tu dau thi nen wipe system+data+cache.

thanhthuy89, 2016-07-27:
Bạn co thuong len gmail ko ? Iien lac cach nao tien va nhanh dc ko? Co bi gi minh hoi

thanhthuy89, 2016-07-27:
Luc nao cai rom minh ko wipe full,moi dung odin up stock lai so stock lỗi,

ngoquang2708, 2016-07-27:
I don't know what cause this issue yet but after sync the repo, the problem is gone.

closed 2016-07-27 by ngoquang2708