ngoquang2708/android_device_samsung_vivalto3gvn #57: Some people reported that their phone cannot goes to deep-sleep

ngoquang2708, 2016-07-27:


koquantam, 2016-07-27:
Check your kernel commits

koquantam, 2016-07-27:
Or some apps are making wakelocks

ngoquang2708, 2016-08-06:
Fixed by commit https://github.com/ngoquang2708/android_packages_apps_CellBroadcastReceiver/commit/8384f1b073e473709683661b42a204adc90259ba.

closed 2016-08-06 by ngoquang2708

pavlaras, 2020-09-20:
> Fixed by commit https://github.com/ngoquang2708/android_packages_apps_CellBroadcastReceiver/commit/8384f1b073e473709683661b42a204adc90259ba.

link deleted. which commit it was ?

ngoquang2708, 2020-09-20:
I cannot recall what I had changed it. Something related to disable CDMA emergency broadcast channels. You can take a look at the code [here](https://github.com/CyanogenMod/android_packages_apps_CellBroadcastReceiver/blob/cm-13.0/src/com/android/cellbroadcastreceiver/CellBroadcastConfigService.java) and figure it out. Sorry if this doesn't help much.