ngoquang2708/android_device_samsung_vivalto3gvn #58: data movil not working

mfenano, 2016-07-27:
hello my model is smg313ml rom version of the original cell is achieved G313MLUBS0APE3_G313MLUUB0AOJ1_G313MLUBU0AOJ1_HOME.tar install your cyanogenmod13 beta 2 without problems the problem is they do not work mobile data 3g wifi only hope can repair this error in its new version

koquantam, 2016-07-28:
It's baseband related problem I guess, because someone reflashed old baseband for SM-G313ML and mobile data works

Anyway, don't make an issue here, unless you're having contributions to our development, and on every issue, please provide radio logcat

ngoquang2708, 2016-07-28:
@mfenano Read through CM13 for Galaxy V thread on XDA forum, you will found workaround for mobile data. Basically, you will have to flash a specific version of your baseband. Detailed information there.

closed 2016-07-28 by ngoquang2708

mfenano, 2016-07-31:
what baseband is recommender for flash this rom?