ngoquang2708/android_device_samsung_vivalto3gvn #60: Auto-reboot to Offline Charging mode when charging with AC power

ngoquang2708, 2016-07-30:


ngoquang2708, 2016-08-06:
Last logcat.
```
6,3114,253469970,-;sec_hal_fg_get_property-14,val->intval = 4520
4,3115,253470007,-;sprdfgu: @sprdfgu_read_vbat_ocv pure...
6,3116,253470022,-;sec_hal_fg_get_property-14,val->intval = 4520
6,3117,253470045,-;sec_hal_fg_get_property-19,val->intval = 0
6,3118,253470067,-;sec_hal_fg_get_property-20,val->intval = 0
4,3119,253470104,-;sprdfgu: @sprdfgu_read_soc delta 0mAh,pclbcnt:25858126
4,3120,253470118,-;sprdfgu: @sprdfgu_read_soc delta capacity 0,full capacity 1500
4,3121,253470133,-;sprdfgu: @sprdfgu_read_soc soc 100,pcapacity 100
4,3122,253470168,-;sprdfgu: @sprdfgu_read_vbat_ocv pure...
4,3123,253470181,-;sprdfgu: @sprdfgu_read_soc cur_ocv =  4520
4,3124,253470196,-;sprdfgu: @sprdfgu_read_soc cur_ocv 4520,capacity_table[0][0] = 4330
6,3125,253470239,-;sec_hal_fg_get_property-43,val->intval = 930
6,3126,253470635,-;sec-battery sec-battery: sec_bat_get_temperature_by_adc: Temp(383), Temp-ADC(1956)
 SUBSYSTEM=platform
 DEVICE=+platform:sec-battery
6,3127,253470659,-;sec-battery sec-battery: sec_bat_get_temperature_by_adc: Temp(650), Temp-ADC(0)
 SUBSYSTEM=platform
 DEVICE=+platform:sec-battery
6,3128,253470680,-;sec-battery sec-battery: sec_bat_get_battery_info:Vnow(4520mV),Inow(0mA),Imax(600mA),SOC(93%),Tbat(383)
 SUBSYSTEM=platform
 DEVICE=+platform:sec-battery
6,3129,253470702,-;sprdchg_get_batt_presence : 0
6,3130,253470721,-;sec-battery sec-battery: sec_bat_battery_cable_check: Battery Connected
 SUBSYSTEM=platform
 DEVICE=+platform:sec-battery
6,3131,253470747,-;sec-battery sec-battery: sec_bat_monitor_work: Status(Charging), mode(Normal), Health(Good), Cable(3), siop_level(100)
 SUBSYSTEM=platform
 DEVICE=+platform:sec-battery
6,3132,253470866,-;sec_hal_fg_get_property-13,val->intval = 4520
3,3133,253470890,-;sec-battery sec-battery: sec_bat_get_property: voltage now(4520)
 SUBSYSTEM=platform
 DEVICE=+platform:sec-battery
6,3134,253471997,-;sec_hal_fg_get_property-13,val->intval = 4520
3,3135,253472031,-;sec-battery sec-battery: sec_bat_get_property: voltage now(4520)
 SUBSYSTEM=platform
 DEVICE=+platform:sec-battery
12,3136,253474925,-;healthd: sec-charger: Unknown power supply type
4,3137,253558424,-;[BACKLIGHT] rt4502_backlight_update_status ==> user_intensity  : 0, tune_level : 0
6,3138,253604210,-;request_suspend_state: sleep (0->3) at 253594599756 (2016-08-06 13:49:36.052225570 UTC)
6,3139,253604262,-;early_suspend: call handlers
6,3140,253604288,-;early_suspend: calling ondemand_gov_early_suspend
6,3141,253604303,-;cpufreq_ondemand: ondemand_gov_early_suspend do nothing
4,3142,253604319,-;-------- suspend ondemand_gov_early_suspend 
6,3143,253604339,-;early_suspend: calling rt4502_backlight_earlysuspend
4,3144,253609466,-;[13:49:36.57489898][BACKLIGHT] earlysuspend
4,3145,253609503,-;-------- suspend rt4502_backlight_earlysuspend 
6,3146,253609522,-;early_suspend: calling sprd_kpled_early_suspend
6,3147,253609536,-;[SPRD_KPLED_INFO] sprd_kpled_early_suspend
4,3148,253609550,-;-------- suspend sprd_kpled_early_suspend 
6,3149,253609569,-;early_suspend: calling ist30xx_early_suspend
4,3150,253609594,-;[ TSP ] ist30xx_power_off()
4,3151,253610763,-;[ TSP ] [TSP] ts_power_enable, 0
4,3152,253669835,-;-------- suspend ist30xx_early_suspend 
6,3153,253669871,-;early_suspend: calling dhd_early_suspend
4,3154,253669885,-;dhd_early_suspend: enter
4,3155,253669906,-;dhd_set_suspend: force extra Suspend setting 
4,3156,253673088,-;dhd_get_suspend_bcn_li_dtim beacon=100 bcn_li_dtim=0 DTIM=2 Listen=10
4,3157,253674895,-;-------- suspend dhd_early_suspend 
6,3158,253674926,-;early_suspend: calling stop_drawing_early_suspend
4,3159,253678190,-;-------- suspend stop_drawing_early_suspend 
6,3160,253678225,-;early_suspend: calling gsp_early_suspend
4,3161,253678240,-;gsp_early_suspend2534
4,3162,253678258,-;-------- suspend gsp_early_suspend 
6,3163,253678277,-;early_suspend: calling sprdfb_early_suspend
6,3164,253678291,-;sprdfb: sprdfb: [sprdfb_early_suspend]
4,3165,253678306,-;esd_det_disable, disable mode : 1 state : 0
6,3166,253678325,-;dispc: [sprdfb_dispc_suspend], dev->enable = 1
4,3167,253678340,-;fb_bl_notifier event=[0x21]
[... 39 lines cut ...]
4,3205,256906891,-;sbuf_read wait interrupted!
4,3206,256906949,-;sbuf_read wait interrupted!
4,3207,256907009,-;sbuf_read wait interrupted!
4,3208,256907068,-;sbuf_read wait interrupted!
4,3209,256907126,-;sbuf_read wait interrupted!
4,3210,256907186,-;sbuf_read wait interrupted!
4,3211,256905300,-;Freezing user space processes ... 
6,3212,256908055,-;active wake lock sprd-sdhci.3-wakelock
6,3213,256908104,-;active wake lock wlan_wd_wake
4,3214,256908148,-;
3,3215,256908197,-;Freezing of tasks aborted after 0.002 seconds (47 tasks refusing to freeze, wq_busy=0):
4,3216,256908239,-;
4,3217,256908279,-;Restarting tasks ... done.
4,3218,256915213,-;*** fw_pm_notify, mode:0x4 done ***
4,3219,256915286,-;*** scxx30_dmcfreq_pm_notifier, event:0x4 ***
6,3220,256915945,-;emc_freq: 192,192,4
6,3221,256916153,-;PM: suspend exit 2016-08-06 13:49:39.364167757 UTC
6,3222,256916216,-;suspend: exit suspend, ret = -16 (2016-08-06 13:49:39.364236710 UTC)
6,3223,256916268,-;active wake lock sprd-sdhci.3-wakelock
6,3224,256916310,-;active wake lock sprd-sdhci.0-wakelock
6,3225,256916352,-;active wake lock mmc1_detect
6,3226,256916393,-;active wake lock wlan_wd_wake
6,3227,256916437,-;active wake lock wlan_rx_wake, time left 50
6,3228,256954743,-;PM: sync done.
```

koquantam, 2016-08-06:
By looking there, I guess the voltage goes too high

ngoquang2708, 2016-08-06:
What would cause that?

koquantam, 2016-08-06:
Idk, ask regulator :joy:

ngoquang2708, 2016-08-21:
Not exits anymore. I replaced my old battery to a new one which causes this issue. It has no problems after switch back to the old one.

closed 2016-08-21 by ngoquang2708