ngoquang2708/android_device_samsung_vivalto3gvn #61: Recent button Spam

PhantomAssassin12, 2016-07-31:
Keeps Spamming the Recent button at 27% of battery and after it shut of.

PhantomAssassin12, 2016-07-31:
@ngoquang2708 do you have this issue?

ngoquang2708, 2016-07-31:
What is "spam" means here? Spamming logcat?

PhantomAssassin12, 2016-07-31:
Keeps pressing the recently button

PhantomAssassin12, 2016-07-31:
I don't touch the screen but still touching the recently button itself.

ngoquang2708, 2016-07-31:
Is you screen broken?

ngoquang2708, 2016-07-31:
I got that too, but my screen is broken so though that was the reason.

PhantomAssassin12, 2016-07-31:
no! my screen is clean don't have any scratches but i have tempered glass protector is this the problem?

PhantomAssassin12, 2016-07-31:
how did you fixed it?

ngoquang2708, 2016-07-31:
I cannot. I just remove /system/usr/keylayout/ist30xx_ts_input.kl and add
qemu.hw.mainkeys=0 to build.prop to use soft-navbar.

PhantomAssassin12, 2016-07-31:
Okay.

PhantomAssassin12, 2016-07-31:
btw can i have your latest build?

closed 2018-06-05 by ngoquang2708