ngoquang2708/android_device_samsung_vivalto3gvn #63: Randomly lost mobile network on Enforcing SELinux mode

ngoquang2708, 2016-08-01:


thanhthuy89, 2016-08-02:
co cach nao fix an recent app ma lag voi xiu moi hien cac ung dung ra ko bac Quang

ngoquang2708, 2016-08-02:
@thanhthuy89 ROM gốc nhanh lắm đó bạn :)

thanhthuy89, 2016-08-02:
minh ko chê... chi la hoi ban co cach lam no nhanj xiu ko thoi

ngoquang2708, 2016-08-02:
Minh khong noi la ban che. Nhung su that la vay ma :D

ngoquang2708, 2016-08-02:
SGV qua yeu.

thanhthuy89, 2016-08-02:
beta3 nay em.youtobe coi ko lag. chi co cai la thanh trang thai keo len xuong giat voi recent app cham.... hoi nay choi 3g nong ran may cai tu khoi dong lai... chua biet phai bi reboot do 3g ko nua

thanhthuy89, 2016-08-09:
giup minh dc ko ?

ngoquang2708, 2016-08-11:
Fix with these commits:
* 61b076160c3d063b9c32724fff2030ad7230e857
* 204a6730fa20ddc12695ca50c66c03c2e14df9eb
* 2c44f6893958a68958caa940e2fb493bab290222
* 13af2e19a7267bf3406925f1db02bb4900f90337
* dbe53794c2228bbed313115acdc7f1f27d226e89
* 851a0fb40f6a7bcca841980081ea2c48b29bff1b

closed 2016-08-11 by ngoquang2708