ngoquang2708/android_device_samsung_vivalto3gvn #66: Random reboot when install the boot.img

hendrulll, 2016-08-06:
The rom is great but with the new selinux enforcing mode, some of my apps cant grant root access, so I decided to install the boot.img, it's fix the app that cant access root because the selinux is now permissive. But I have new problem, random reboot, it never happen if I dont install the boot.img

ngoquang2708, 2016-08-06:
Can I have kmsg log?

ngoquang2708, 2016-08-06:
Please take it with SELinux Enforcing.

2016-08-06 22:51 GMT+07:00 Quang Ngô <ngoquang2708@gmail.com>:

hendrulll, 2016-08-06:
What is kmsg log? how can I do that? Btw sorry I'm newbie

ngoquang2708, 2016-08-06:
Run adb shell cat /dev/kmsg > log.txt, open your app, do stuffs, then press
ctrl+c in your terminal. Take your log.txt here.

hendrulll, 2016-08-06:
It says cant create log.txt: read only file system

ngoquang2708, 2016-08-06:
OK, run this:

adb shell dmesg

hendrulll, 2016-08-06:
sh: adb: not found

ngoquang2708, 2016-08-06:
OK, if you run Terminal in you phone, do this.
Open Terminal
cd /storage/emulated/0/
su
dmesg > log.txt

2016-08-06 23:13 GMT+07:00 Quang Ngô <ngoquang2708@gmail.com>:

ngoquang2708, 2016-08-06:
That log.txt will be iin your internal storage.

2016-08-06 23:19 GMT+07:00 Quang Ngô <ngoquang2708@gmail.com>:

hendrulll, 2016-08-06:
[log.txt](https://github.com/ngoquang2708/android_device_samsung_vivalto3gvn/files/405078/log.txt)

ngoquang2708, 2016-08-06:
Thank you. I will investigate it.

hendrulll, 2016-08-06:
Yea, hope you can fix it soon. Thanks for always available bruh

thanhthuy89, 2016-08-07:
Use selinux mod changer.apk can chance between enforcing and permissive

ngoquang2708, 2016-08-11:
Not exists any more in my latest build. I will release it soon.

closed 2016-08-11 by ngoquang2708