ngoquang2708/android_device_samsung_vivalto3gvn #67: Pokemon Go Forced Close

PhantomAssassin12, 2016-08-07:
@ngoquang2708 
The pokemon Go is closing after I login without any box that says pokemon has force closed.
[LogFile Here.](https://github.com/ngoquang2708/android_device_samsung_vivalto3gvn/files/405241/logfile.txt)

ngoquang2708, 2016-08-07:
Our device doesn't meet the requirement to able to run PG.

koquantam, 2016-08-07:
Pokemon Go is actually can be played with our phone, but stock and CM12.1 only (I have seen that in my FB group)

PhantomAssassin12, 2016-08-07:
I Flash the stock but still no luck.

bagaskara815, 2016-08-07:
im using GLtools in CM11 but still Lag and then crash

koquantam, 2016-08-07:
@bagaskara815 
GLtools is just a placebo software for our device. Since our GPU was just single-core with a very low speed (312MHz), you can't except any improvements with GLtools, because it just do some hacks with OpenGL rendering module (libGLESv2 and libGLESv3), not making "magics" with our phone

bagaskara815, 2016-08-08:
@koquantam thanks for you explanation

ngoquang2708, 2016-08-08:
So this issue is not related to this ROM but the hardware it self, confirmed with @PhantomAssassin12 report.

closed 2016-08-08 by ngoquang2708