ngoquang2708/android_device_samsung_vivalto3gvn #70: Cannot switch mobile data without reboot

ngoquang2708, 2016-08-14:
There are changes in ```DcSwitchStateMachine``` since CM-12.1. I think those changes cause this issue but I am not sure. Point it out here to have a clue for more investigation.

koquantam, 2016-08-14:
You mean soft reboot?

ngoquang2708, 2016-08-14:
No soft reboot but I must reboot to switch mobile data.

koquantam, 2016-08-14:
Sorry for not asking properly, but I mean do you have to do a soft reboot to switch mobile data SIM or a hard reboot?

This was existed since CM11, not CM12.1

ngoquang2708, 2016-08-14:
AirplaneMode and HardReboot works, SoftReboot does'nt. It means I have to
fully reset the modem.

thanhthuy89, 2016-08-17:
bác quang oi... toi da test cai nay la co 100%.. khi toi choi game online 3g lien tuc bi doi thang 2g.... stock thi ko co.. mong bac fix som

ngoquang2708, 2016-08-21:
Fixed by commit https://github.com/ngoquang2708/android_device_samsung_vivalto3gvn/commit/555ab6cfbb052d2a42028b3443644dee3083149c.

closed 2016-08-21 by ngoquang2708