ngoquang2708/android_device_samsung_vivalto3gvn #72: Cannot built the kernel with "arm-linux-androideabi-" toolchain

ngoquang2708, 2016-08-28:


koquantam, 2016-08-28:
What does the compiler gives?

ngoquang2708, 2016-08-28:
I can built with cm-13, but there is some linker warning and I cannot boot
that kernel. I cannot build the kernel at all in cm-14.0. It throws errors
when parsing __PASTE macro in include/linux/compiler-gcc4.h which caused
when building sound/soc/sprd/dai/sprd-pcm.c, when expand MODULE_ALIAS macro.

koquantam, 2016-08-28:
LOL try using other compiler, like Linaro 5.1 or sth
Or use my compiler-gcc.h in my grandprimeve3g kernel source, as in 3.10.102, compiler-gccX are deprecated

ngoquang2708, 2016-08-28:
OK, I will test it later. Using arm-eabi-4.8 currently.

ngoquang2708, 2016-09-02:
Fix by commit https://github.com/ngoquang2708/android_kernel_samsung_vivalto3gvn/commit/563da18bca372b290b2247975161836a203b3efe.

closed 2016-09-02 by ngoquang2708