ngoquang2708/android_device_samsung_vivalto3gvn #76: No mobile data

ngoquang2708, 2016-09-04:


fadhilm293, 2016-09-17:
Sir can i test Aosp 6.0 please.

ngoquang2708, 2016-09-18:
@fadhilm293 Here you are http://www.mediafire.com/download/12okiz199o29v2v/aosp-6.0-vivalto3gvn.zip.

ngoquang2708, 2016-09-18:
@fadhilm293 Install it with this https://github.com/ngoquang2708/android_device_samsung_vivalto3gvn/files/462135/twrp2.img.zip.

fadhilm293, 2016-09-18:
Thanks sir,Btw is there any other bug?

ngoquang2708, 2016-09-18:
@fadhilm293 You tell me :)

fadhilm293, 2016-09-18:
List of bug that i found:
- Baseband unknow
- systemui and com.android.phone always force close
- some application force close
- can't detect sim card.

ngoquang2708, 2016-09-18:
Yeah, that is it! The goal is not to build AOSP 6 but 7.

fadhilm293, 2016-09-18:
Okay,but please fix bug on this rom because this rom very smoothly.

ngoquang2708, 2016-09-18:
I would do if I have time. But I don't. So sorry that I can't.

fadhilm293, 2016-09-18:
Okay,please tell me if the bug has been fixed,you are the best.

fadhilm293, 2016-09-18:
If not,please let me know how to fix that bug.

ngoquang2708, 2016-09-18:
I have not identified the causes of those bugs so I can't tell you.

ngoquang2708, 2016-09-22:
Fixed by these commits: https://github.com/ngoquang2708/android_device_samsung_vivalto3gvn/commit/40b4629186d58ccd540b4d5f44b0aa06bb4b5179 https://github.com/ngoquang2708/android_device_samsung_vivalto3gvn/commit/3773ed9e1e46515feafc79315265ca1c63f16612.

closed 2016-09-22 by ngoquang2708

thanhthuy89, 2016-09-23:
i can't flash aosp 6.0 by twrp ( 3.0.2.0 & 2.8.5.0) error . please repair aosp 6.0

fadhilm293, 2016-09-24:
I try and find bug on aosp6
Among others :
- Can't play video videos in stock gallery and Third-party applications
- The camera upside down after taking a picture.

Sorry for bad English.

fadhilm293, 2016-09-24:
Sorry miss, I tried it on AOSP Beta 1.

ngoquang2708, 2016-09-24:
@fadhilm293 Thank you. I will check it.

fadhilm293, 2016-09-25:
Sir,I found another bug.

- Setup wizard has force close after Sim card detected and set up sim card.

Sorry for bad English.😁

fadhilm293, 2016-09-25:
My Phone is SM-G313HZ.

ngoquang2708, 2016-09-25:
I have the same variant as yours, but don't encounter those issues. Please
provide logcat after each of those issues happened, then open new issue (on
github) for each of them and attach logcat there. I would greatly
appreciate.

fadhilm293, 2016-10-05:
@ngoquang2708 Here is logcat 
http://upfile.mobi/VKlTgEJwLIv

fadhilm293, 2016-10-06:
@ngoquang2708 please give me file to fix video codec and Hw rendering,because if i'am use Sw rendering,its so sluggish for playing video.

ngoquang2708, 2016-10-06:
@fadhilm293 N ate all of my time, I do not have time for M. Sorry.