ngoquang2708/android_device_samsung_vivalto3gvn #78: Soft-reboot when screen is turning off

ngoquang2708, 2016-09-05:


thanhthuy89, 2016-09-05:
build oasp 7.0 nougat for galaxy V ?

fadhilm293, 2016-09-17:
Sir,Has there been progress this rom ?

ngoquang2708, 2016-09-17:
WIP.

ngoquang2708, 2016-09-28:
Graphics codecs in ```ColorFade.java``` causes this crash.

ngoquang2708, 2016-09-28:
Workaround in this commit https://github.com/ngoquang2708/android_frameworks_base/commit/5da2bacc98328c52fc3d824910e77e2c18471de4.

thanhthuy89, 2016-09-28:
can you share me to fix video codec in youtobe app and gallary ?

ngoquang2708, 2016-09-28:
@thanhthuy89 I think you should CM13. It is just the same as AOSP6 with more features. Do not think that AOSP6 is faster then CM13, the speed is the same when you have installed Gapps and other apps.

koquantam, 2016-09-28:
He wants to use AOSP 6.0 because CM13 has UI bug, which sometimes cause lags with transition animations

ngoquang2708, 2016-10-11:
Partially fixed by commit https://github.com/ngoquang2708/android_device_samsung_vivalto3gvn/commit/f334e55735be5805696f5ddeaf6ddc97fae6d5ce.

thanhthuy89, 2016-10-11:
cab you share zip file fix UI bug cm13?

ngoquang2708, 2016-10-11:
@thanhthuy89 What UI bug?

thanhthuy89, 2016-10-11:
UI bug, which sometimes cause lags with transition animations

thanhthuy89, 2016-10-17:
can you share for me aosp 7.0 fixed soft reboot ?

fadhilm293, 2016-10-17:
@ngoquang2708

Can you share a file to fix the camera in aosp7?
because I use it for everyday.

ngoquang2708, 2016-10-17:
@thanhthuy89 I do not have time to build AOSP 7. Sorry about that.
@fadhilm293 People are working on it. See http://review.cyanogenmod.org/#/c/165008/.

ngoquang2708, 2016-11-19:
It is gone!

closed 2016-11-19 by ngoquang2708