ngoquang2708/android_device_samsung_vivalto3gvn #82: wifi: connect to AP cause boot loop

ngoquang2708, 2016-09-28:


ngoquang2708, 2016-10-01:
```10-01 21:02:08.016  2121  2221 E WifiHW  : Unable to open connection to supplicant on "@android:wpa_wlan0": Permission denied```

ngoquang2708, 2016-10-01:
```
10-01 21:25:38.697   199   581 V IdletimerController: runCmd(/system/bin/ip6tables -w -t raw -A idletimer_raw_PREROUTING -i wlan0 -j IDLETIMER --timeout 15 --label 1 --send_nl_msg 1) res_ipv4=1, res_ipv6=1
10-01 21:25:38.707   525   609 E ConnectivityService: Exception in setupDataActivityTracking java.lang.IllegalStateException: command '42 idletimer add wlan0 15 1' failed with '400 42 Failed to add interface'
```

ngoquang2708, 2016-10-01:
If I connect to AP right in CMSetupWizard, bootloop would gone! So, Is CMSetupWizard issue???

koquantam, 2016-10-01:
Maybe, and what's the problem of CMSW with you?

ngoquang2708, 2016-10-01:
Disconnect then reconnect WiFi, bootloop again!

ngoquang2708, 2016-10-17:
Latest updates from CM repos fixed this issue.

ngoquang2708, 2016-10-29:
This issue re-appear again only when I enabled dex-preopt. The one that crash is org.cyanogenmod.cmparts. It crash right after connected to the WiFi.

closed 2018-06-05 by ngoquang2708