ngoquang2708/android_device_samsung_vivalto3gvn #89: bluetooth only receive,  can't send any file

thanhthuy89, 2016-10-31:


ngoquang2708, 2016-11-21:
Logcat:
```
11-21 18:16:40.748   203   308 W audio_hw_primary: [out_set_parameters], kvpairs=routing=2 devices:0x2 mode:0 
11-21 18:16:40.748   203   308 W audio_hw_primary: [out_set_parameters],after str_parms_get_str,val(0x2) 
11-21 18:16:40.748   203   308 W audio_hw_primary: the same devices(0x2) with val(0x2) val is zero...
11-21 18:16:40.748   203   308 W audio_hw_primary: out_set_parameters out...call_start:0
11-21 18:16:40.751   203   308 D audio_hw_primary: start output stream out->is_voip is 0, in
11-21 18:16:40.751   203   308 I audio_hw_primary: select_devices_signal starting...
11-21 18:16:40.751   203   308 I audio_hw_primary: select_devices_signal finished.
11-21 18:16:40.751   203   308 W audio_hw_primary: open s_tinycard in
11-21 18:16:40.752   203   308 W audio_hw_primary: open s_tinycard successfully
11-21 18:16:40.753   203   304 I audio_hw_primary: Not to change mute: 0
11-21 18:16:40.753   203   304 I audio_hw_primary: Changing out_devices: from (0xfffffffd) to (0x00000002)
11-21 18:16:40.753   203   304 I audio_hw_primary: Changing in_devices: from (0x00000000) to (0x00000000)
11-21 18:16:40.754   203   304 I audio_hw_primary: Set 'SPKL Mixer DACLSPKL Switch'.0 to 1
11-21 18:16:40.754   203   304 I audio_hw_primary: Set 'SPKL Mixer DACRSPKL Switch'.0 to 1
11-21 18:16:40.754   203   304 I audio_hw_primary: Set 'Speaker Function'.0 to 1
11-21 18:16:40.754   203   304 I audio_hw_primary: Set 'Mic Function'.0 to 0
11-21 18:16:40.754   203   304 I audio_hw_primary: Set 'ADCL Mixer MainMICADCL Switch'.0 to 0
11-21 18:16:40.755   203   304 I audio_hw_primary: Set 'ADCR Mixer MainMICADCR Switch'.0 to 0
11-21 18:16:40.755   203   304 I audio_hw_primary: Set 'Aux Mic Function'.0 to 0
11-21 18:16:40.755   203   304 I audio_hw_primary: Set 'ADCL Mixer AuxMICADCL Switch'.0 to 0
11-21 18:16:40.755   203   304 I audio_hw_primary: Set 'ADCR Mixer AuxMICADCR Switch'.0 to 0
11-21 18:16:40.755   203   304 I audio_hw_primary: Set 'HP Mic Function'.0 to 0
11-21 18:16:40.755   203   304 I audio_hw_primary: Set 'ADCL Mixer HPMICADCL Switch'.0 to 0
11-21 18:16:40.755   203   304 I audio_hw_primary: Set 'ADCR Mixer HPMICADCR Switch'.0 to 0
11-21 18:16:40.755   203   304 I audio_hw_primary: Set 'SPKL Mixer ADCLSPKL Switch'.0 to 0
11-21 18:16:40.755   203   304 I audio_hw_primary: Set 'SPKL Mixer ADCRSPKL Switch'.0 to 0
11-21 18:16:40.755   203   304 I audio_hw_primary: Set 'HPL Mixer ADCRHPL Switch'.0 to 0
11-21 18:16:40.755   203   304 I audio_hw_primary: Set 'HPR Mixer ADCLHPR Switch'.0 to 0
11-21 18:16:40.755   203   304 I audio_hw_primary: Set 'Line Function'.0 to 0
11-21 18:16:40.755   203   304 I audio_hw_primary: Set 'ADCL Mixer AILADCL Switch'.0 to 0
11-21 18:16:40.755   203   304 I audio_hw_primary: Set 'ADCR Mixer AIRADCR Switch'.0 to 0
11-21 18:16:40.755   203   304 I audio_hw_primary: Set 'HeadPhone Function'.0 to 0
11-21 18:16:40.755   203   304 I audio_hw_primary: Set 'HPL Mixer DACLHPL Switch'.0 to 0
11-21 18:16:40.755   203   304 I audio_hw_primary: Set 'HPR Mixer DACRHPR Switch'.0 to 0
11-21 18:16:40.755   203   304 I audio_hw_primary: Set 'Earpiece Function'.0 to 0
11-21 18:16:40.755   203   304 I vb_effect: s_cur_out_devices(0x00000002), s_cur_in_devices(0x00000000)
11-21 18:16:40.755   203   304 I vb_effect: profile is Handsfree, ret=0
11-21 18:16:40.755   203   304 W audio_hw_primary: GetAudio_gain_by_devices, outmode:3 inmode:3,indev:0x0, outdev:0x2
11-21 18:16:40.755   203   304 W audio_hw_primary: vb_control_parameters.c GetAudio_outpga_nv, dac_pga_gain_l:0x801807c7 adc_pga_gain_l:0x0 fm_pga_gain_l:0x227 fm_pga_gain_r:0x227 pa_config:0x330, 0x330(fm),  vol_level:0x1 
11-21 18:16:40.755   203   304 W audio_hw_primary: GetAudio_inpga_nv, adc_pga_gain_l:0x8f23 device:0x0 vol_level:0x1
11-21 18:16:40.755   203   304 D audio_pga: 'speaker-l' apply
11-21 18:16:40.755   203   304 I audio_pga: 'SPKL Playback Volume' set to 7
11-21 18:16:40.756   203   304 I audio_pga: 'DACL Playback Volume' set to 7
11-21 18:16:40.756   203   304 I audio_pga: 'VBC DACL DG Set' set to 24
11-21 18:16:40.756   203   304 I audio_pga: 'VBC DACL DG Switch' set to 1
11-21 18:16:40.756   203   304 D audio_pga: 'speaker-r' apply
11-21 18:16:40.756   203   304 I audio_pga: 'SPKR Playback Volume' set to 7
11-21 18:16:40.756   203   304 I audio_pga: 'DACR Playback Volume' set to 7
11-21 18:16:40.756   203   304 I audio_pga: 'VBC DACR DG Set' set to 24
11-21 18:16:40.756   203   304 I audio_pga: 'VBC DACR DG Switch' set to 1
11-21 18:16:40.756   203   304 W audio_hw_primary: SetAudio_gain_by_devices out, out_devices:0x2, in_devices:0x0
11-21 18:16:40.756   203   304 I audio_hw_primary: stream_routing_thread looping done.
11-21 18:16:40.756   203   304 I audio_hw_primary: stream_routing_thread looping now...
11-21 18:16:40.840   950   950 W ContextImpl: Calling a method in the system process without a qualified user: android.app.ContextImpl.sendBroadcast:877 android.content.ContextWrapper.sendBroadcast:421 com.android.settings.bluetooth.DevicePickerFragment.sendDevicePickedIntent:152 com.android.settings.bluetooth.DevicePickerFragment.onDevicePreferenceClick:119 com.android.settings.bluetooth.DeviceListPreferenceFragment.onPreferenceTreeClick:148 
11-21 18:16:40.864  2587  2587 D BluetoothOppReceiver: Enter - onReceive for intent:android.bluetooth.devicepicker.action.DEVICE_SELECTED
11-21 18:16:40.898   950   950 D LocalBluetoothManager: setting foreground activity to null
[... cut ...]
11-21 18:16:47.645   492   530 I WindowManager: Destroying surface Surface(name=Toast) called by com.android.server.wm.WindowStateAnimator.destroySurface:2033 com.android.server.wm.WindowStateAnimator.destroySurfaceLocked:884 com.android.server.wm.WindowState.destroyOrSaveSurface:2095 com.android.server.wm.WindowSurfacePlacer.performSurfacePlacementInner:429 com.android.server.wm.WindowSurfacePlacer.performSurfacePlacementLoop:232 com.android.server.wm.WindowSurfacePlacer.performSurfacePlacement:180 com.android.server.wm.WindowManagerService$H.handleMessage:8271 android.os.Handler.dispatchMessage:102
```

Verevka, 2017-09-17:
How did you fix this?

ngoquang2708, 2017-09-17:
It is not fixed.

remilia15, 2017-09-17:
Buggy Bluetooth is fixed on latest N then