ngoquang2708/android_device_samsung_vivalto3gvn #95: surfaceflinger crashes

ngoquang2708, 2018-05-29:
```
*** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
LineageOS Version: '15.1-20180528-UNOFFICIAL-vivalto3gvn'
Build fingerprint: 'samsung/lineage_vivalto3gvn/vivalto3gvn:8.1.0/OPM2.171019.029.B1/b94c87449e:eng/test-keys'
Revision: '0'
ABI: 'arm'
pid: 152, tid: 152, name: surfaceflinger  >>> /system/bin/surfaceflinger <<<
signal 6 (SIGABRT), code -6 (SI_TKILL), fault addr --------
Abort message: 'EGLContext creation failed'
    r0 00000000  r1 00000098  r2 00000006  r3 00000008
    r4 00000098  r5 00000098  r6 be9ce4fc  r7 0000010c
    r8 00000000  r9 00000000  sl 60000014  fp be9cea9c
    ip a7cc2240  sp be9ce4e8  lr a82399d7  pc a8233408  cpsr 200f0030
    d0  6c696166206e6f69  d1  7461657263207478
    d2  6665725f6f74753a  d3  4c47452068736520
    d4  65766974616e5f74  d5  5f746e65696c635f
    d6  4520726566667562  d7  4f52444e415f4c47
    d8  0000000000000000  d9  0000000000000000
    d10 0000000000000000  d11 0000000000000000
    d12 0000000000000000  d13 0000000000000000
    d14 0000000000000000  d15 0000000000000000
    d16 2e74736973726570  d17 2e6761742e676f6c
    d18 0000000000fe4f15  d19 0000000000000000
    d20 0000000800000008  d21 0000000800000008
    d22 0000000400000004  d23 0000000400000004
    d24 0000240100000401  d25 0000640100004401
    d26 0000240000000400  d27 0000640000004400
    d28 0000000000000000  d29 0000000000000000
    d30 0000000000000000  d31 0000000000000000
    scr 20000010

backtrace:
    #00 pc 0001a408  /system/lib/libc.so (abort+63)
    #01 pc 000065c3  /system/lib/liblog.so (__android_log_assert+154)
    #02 pc 0003ea61  /system/lib/libsurfaceflinger.so (android::RenderEngine::create(void*, int, unsigned int)+824)
    #03 pc 00041c6b  /system/lib/libsurfaceflinger.so (android::SurfaceFlinger::init()+414)
    #04 pc 0000205f  /system/bin/surfaceflinger (main+394)
    #05 pc 00077489  /system/lib/libc.so (__libc_init+48)
    #06 pc 00001df4  /system/bin/surfaceflinger (_start_main+88)

stack:
         be9ce4a8  ffffffdf
         be9ce4ac  00000000
         be9ce4b0  be9cea9c
         be9ce4b4  e34e88f1
         be9ce4b8  00000098
         be9ce4bc  a9074181  /system/bin/surfaceflinger (sigprocmask+184)
         be9ce4c0  00000007
         be9ce4c4  be9ce4f8
         be9ce4c8  ffffffdf
         be9ce4cc  e34e88f1
         be9ce4d0  00000098
         be9ce4d4  00000098
         be9ce4d8  be9ce4fc
         be9ce4dc  a7cc2240  [anon:libc_malloc]
         be9ce4e0  00000000
         be9ce4e4  a82333fd  /system/lib/libc.so (abort+52)
    #00  be9ce4e8  00000001
         be9ce4ec  a7f2a638  /system/lib/libsurfaceflinger.so
         be9ce4f0  0000000f
         be9ce4f4  be9ce530
         be9ce4f8  0000001b
         be9ce4fc  ffffffdf
         be9ce500  a7f2a638  /system/lib/libsurfaceflinger.so
         be9ce504  be9ce520
         be9ce508  00000000
         be9ce50c  a7cc2240  [anon:libc_malloc]
         be9ce510  00000001
         be9ce514  a89145c7  /system/lib/liblog.so (__android_log_assert+158)
    #01  be9ce518  a7f2b6c8  /system/lib/libsurfaceflinger.so
[... 2334 lines cut ...]
01-01 00:00:13.971   152   152 D vndksupport: Loading /vendor/lib/hw/android.hardware.graphics.mapper@2.0-impl.so from current namespace instead of sphal namespace.
01-01 00:00:13.973   152   152 D vndksupport: Loading /vendor/lib/hw/gralloc.scx15.so from current namespace instead of sphal namespace.
01-01 00:00:13.978   152   152 D vndksupport: Loading /vendor/lib/hw/android.hardware.graphics.allocator@2.0-impl.so from current namespace instead of sphal namespace.
01-01 00:00:13.983   152   152 D vndksupport: Loading /vendor/lib/hw/gralloc.scx15.so from current namespace instead of sphal namespace.
01-01 00:00:14.000   152   152 D vndksupport: Loading /vendor/lib/hw/gralloc.scx15.so from current namespace instead of sphal namespace.
01-01 00:00:14.002   152   152 I SPRDHWComposer: DisplayPlane createPlaneBuffer phy addr:0x9faee000, size:1536000
01-01 00:00:14.099   152   152 D vndksupport: Loading /vendor/lib/hw/gralloc.scx15.so from current namespace instead of sphal namespace.
01-01 00:00:14.116   152   152 I SPRDHWComposer: DisplayPlane createPlaneBuffer phy addr:0x9fc95000, size:1536000
01-01 00:00:14.117   152   152 I [Gralloc]: dither close ID 1
01-01 00:00:14.117   152   152 I SurfaceFlinger: Using composer version 1.3
01-01 00:00:14.117   152   152 I SPRDHWComposer: getDisplayAttributes: disp:0 vsync_period:16666389
01-01 00:00:14.117   152   152 I SPRDHWComposer: getDisplayAttributes: disp:0 width:480
01-01 00:00:14.118   152   152 I SPRDHWComposer: getDisplayAttributes: disp:0 height:800
01-01 00:00:14.118   152   152 I SPRDHWComposer: getDisplayAttributes: disp:0 xdpi:234.461531
01-01 00:00:14.118   152   152 I SPRDHWComposer: getDisplayAttributes: disp:0 ydpi:233.563219
01-01 00:00:14.118   152   152 D SPRDHWComposer: External Display Device is not connected
01-01 00:00:14.118   152   152 D SPRDHWComposer: External Display Device is not connected
01-01 00:00:14.118   152   152 I SurfaceFlinger: EGL information:
01-01 00:00:14.118   152   152 I SurfaceFlinger: vendor    : Android
01-01 00:00:14.119   152   152 I SurfaceFlinger: version   : 1.4 Android META-EGL
01-01 00:00:14.119   152   152 I SurfaceFlinger: extensions: EGL_KHR_get_all_proc_addresses EGL_ANDROID_presentation_time EGL_KHR_swap_buffers_with_damage EGL_ANDROID_get_native_client_buffer EGL_ANDROID_front_buffer_auto_refresh EGL_ANDROID_get_frame_timestamps 
01-01 00:00:14.119   152   152 I SurfaceFlinger: Client API: OpenGL_ES
01-01 00:00:14.119   152   152 I SurfaceFlinger: EGLSurface: 8-8-8-8, config=0x60000014
01-01 00:00:14.119   152   152 F SurfaceFlinger: EGLContext creation failed
```

remilia15, 2018-05-30:
Don't leave egl libs on vendor partition. They aren't VNDK supported

remilia15, 2018-05-30:
Oh wait, did you notice these?
```
01-01 00:00:13.874   152   152 E SurfaceFlinger: Couldn't set SCHED_FIFO for SFEventThread
01-01 00:00:13.874   152   152 E SurfaceFlinger: Couldn't set SCHED_FIFO for EventThread
```

ngoquang2708, 2018-05-30:
cm-14.1 has that too. In Oreo, surfaceflinger has successfully load hwc gralloc. But it failed to initialize egl context. I am stuck at that. My only clue is that difference binder device cause that. Not sure.Off topic. Are you stuck at brining ril to work? May be that because of new pthread in O. Like my issue on phoneserver, I patched bionic and make phoneserver not to crash. Hope that help.
null

ngoquang2708, 2018-05-30:
Well, look at the near end of the log, it seem that those egl libs are loaded successfully.

"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "37567f93-e2a7-4e2a-ad37-a9160fc62647",
"title": "Re: [ngoquang2708/android_device_samsung_vivalto3gvn] surfaceflinger crashes (#95)",
"sections": [
{
"text": "",
"activityTitle": "**Remilia Scarlet**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@remilia15",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"ngoquang2708/android_device_samsung_vivalto3gvn\",\n\"issueId\": 95,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"ngoquang2708/android_device_samsung_vivalto3gvn\",\n\"issueId\": 95\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/ngoquang2708/android_device_samsung_vivalto3gvn/issues/95#issuecomment-393022303"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 340098561\n}"
}
],
"themeColor": "26292E"
}

ngoquang2708, 2018-05-30:
Ah, I got it. libMali use dlopen to load egl libs in which it use absolute path to /system. I have patched libMali to load it from /vendor. Currently surfaceflinger is loaded egl successfully, but still crash.
https://gist.github.com/ngoquang2708/5b51e014548ea6ab9f968cad87fa17aa

remilia15, 2018-05-30:
Well actually RIL is... yeah, a big issue for me
RIL on sc8830 couldn't get signals (so can't make calls too) but can receive and answer calls, messages. I was trying to disassembly stock libril so see which RIL signals implementation is being used on it

Now I can't test anything tho, as my PC is dead and my lappy's RAM couldn't handle compiling O

remilia15, 2018-05-30:
I remember there's a flag for disabling fatal throw on bionic but seems LOS guys removed it, for security reasons

remilia15, 2018-06-05:
Did you manage to boot with HIDL graphics HAL?

ngoquang2708, 2018-06-05:
No HIDL. With it, mali failed to validate the memory allocated by gralloc. Maybe difference memory location since HIDL use a seperated process to load gralloc HAL.

remilia15, 2018-06-05:
We can use our custom HIDL gralloc implementation. Just a hint

ngoquang2708, 2018-06-05:
How is that possible?

remilia15, 2018-06-05:
HIDL gralloc works as a normal HAL, in vendor partition (as we don't have, it's our vendor folder in system)

remilia15, 2018-06-05:
As long as we have our HIDL services called via HIDL API, we can implement custom HIDL stuffs

remilia15, 2018-06-05:
An example for HIDL service is this
https://github.com/LineageOS/android_hardware_interfaces/blob/lineage-15.1/graphics/allocator/2.0/default/service.cpp

remilia15, 2018-06-05:
Actually we have those graphics HIDL HALs run as passthrough by default
What confusing me is how I got out of memory error with gralloc. I'm thinking about implementing CMA from kernel side

ngoquang2708, 2018-06-05:
That would be a hard job!

remilia15, 2018-06-11:
I found this commit. Maybe this will help with our galloc
https://gerrit.aicp-rom.com/c/AICP/frameworks_av/+/63128/4/media/libstagefright/SurfaceUtils.cpp#b113

ngoquang2708, 2018-06-23:
@remilia15 What is the correct way to setup HIDL HALs? Currently I set it up like this:
```
android.hardware.graphics.allocator@2.0-impl
android.hardware.graphics.allocator@2.0-service
android.hardware.graphics.composer@2.1-impl
android.hardware.graphics.composer@2.1-service
android.hardware.graphics.mapper@2.0-impl
```

ngoquang2708, 2018-06-23:
What is passthrough and binderized HIDL HALs. I have read about them at source.android.com but still confuse.

remilia15, 2018-06-23:
All HALs are passthrough if they don't have a service package (e.g `android.hardware.graphics.composer@2.1-service`)

ngoquang2708, 2018-06-23:
OK. Thank you!

remilia15, 2018-07-17:
After tons of experiments and failures, I just realized (blame my noobness) that we need custom implementation of `gralloc_alloc_framebuffer_locked` and cannot rely on private_handle_t member, since we use `shared_fd` for ION, but the fb device doesn't need it as you said, and mapper treats it as framebuffer fd so as a result, `native_handle_clone` returns NULL, and HIDL gralloc couldn't initialize (there's a simple logic that libui sends allocate request to HIDL gralloc, and HIDL gralloc sends buffer request to mapper)

If we couldn't do this, the camera will die forever

ngoquang2708, 2018-07-22:
OMG. I just realized how dump I am until now. Now I know what is binderized HALs vs passthrough HALs. Smh.

remilia15, 2018-07-22:
Btw can you boot with HWC? Mine just dead with NPE, and WiFi display needs HWC to work

ngoquang2708, 2018-07-22:
Not yet.

ngoquang2708, 2018-07-23:
@remilia15 Did you notice this. When running binderized composer, this log appear twice
```
07-23 22:24:32.707   178   382 I [Gralloc]: using (fd=11)
07-23 22:24:32.707   178   382 I [Gralloc]: id           = sprdfb
07-23 22:24:32.707   178   382 I [Gralloc]: xres         = 480 px
07-23 22:24:32.707   178   382 I [Gralloc]: yres         = 800 px
07-23 22:24:32.707   178   382 I [Gralloc]: xres_virtual = 480 px
07-23 22:24:32.707   178   382 I [Gralloc]: yres_virtual = 1600 px
07-23 22:24:32.707   178   382 I [Gralloc]: bpp          = 32
07-23 22:24:32.707   178   382 I [Gralloc]: r            =  0:8
07-23 22:24:32.707   178   382 I [Gralloc]: g            =  8:8
07-23 22:24:32.707   178   382 I [Gralloc]: b            = 16:8
07-23 22:24:32.707   178   382 I [Gralloc]: width        = 52 mm (234.461533 dpi)
07-23 22:24:32.707   178   382 I [Gralloc]: height       = 87 mm (233.563217 dpi)
07-23 22:24:32.707   178   382 I [Gralloc]: refresh rate = 60.00 Hz

```

but with difference pid (and framebuffer fd). They are sufaceflinger and android.hardware.graphics.composer@?-service. So surfaceflinger does not use android.hardware.graphics.composer@?-service???

ngoquang2708, 2018-07-23:
Digging surfaceflinger code :D

remilia15, 2018-07-23:
It must be in use, because HIDL paths is preferred more than legacy

remilia15, 2018-07-23:
I feel so fucking weird now, because HIDL allocator runs with patched libui on your scx15 :/

ngoquang2708, 2018-07-23:
I removed patched libui. Now I got why there is two opening of the previous log. They are bootanimation and surfacelinger. all binderized hal and bootanimation is not crash, just surfaceflinger keep crashing on this
```
07-23 22:24:49.551   521   521 W GrallocMapperPassthrough: buffer descriptor with invalid usage bits 0x400
07-23 22:24:49.551   521   521 E GraphicBufferAllocator: Failed to allocate (480 x 800) layerCount 1 format 1 usage 1e02: 5
07-23 22:24:49.551   521   521 E BufferQueueProducer: [FramebufferSurface] dequeueBuffer: createGraphicBuffer failed
07-23 22:24:49.551   521   521 E [EGL-ERROR]: void __egl_platform_dequeue_buffer(egl_surface*):1854: failed to dequeue buffer from native window 0xa62ff808; err = -12, buf = 0x0,max_allowed_dequeued_buffers 1
07-23 22:24:49.551   521   521 E libEGL  : eglMakeCurrent:1062 error 3003 (EGL_BAD_ALLOC)
07-23 22:24:49.551   521   521 E SurfaceFlinger: DisplayDevice::makeCurrent on default display failed. Aborting.

```

remilia15, 2018-07-23:
You know that the goddamn fb in gralloc is fucking with `shared_fd`
But how HIDL allocator and composer can run with patched libui o.O

ngoquang2708, 2018-07-23:
I think they are running but there is no allocation request to them so they keep alive :D

remilia15, 2018-07-24:
I just looked at the patch again, and it requires mapper to run allocator service o.O
The allocator service is registered but doesn't run, so you were right

remilia15, 2018-07-24:
@ngoquang2708 I just noticed an interesting thing with HIDL mapper used with patched libui: It doesn't even touch framebuffer stuffs at all

remilia15, 2018-07-24:
Oh my bad. I wasn't using allocator with mapper that time so mapper won't touch framebuffer device

ngoquang2708, 2018-07-24:
I am thinking of a way to pass the -1 fd to mali. We need to find the way android pass the native_handle (private_handle) to mali. In gralloc, we can fake that fd say 0 so that native_handle_clone will success, then before passing that cloned handle to mali, we set it back to -1 so that we don't confuse mali.

ngoquang2708, 2018-07-25:
If I run with the following config:
```
    <hal format="hidl">
        <name>android.hardware.graphics.allocator</name>
        <transport>**hwbinder**</transport>
        <version>2.0</version>
        <interface>
            <name>IAllocator</name>
            <instance>default</instance>
        </interface>
    </hal>
    <hal format="hidl">
        <name>android.hardware.graphics.composer</name>
        <transport arch="32">passthrough</transport>
        <version>2.1</version>
        <interface>
            <name>IComposer</name>
            <instance>default</instance>
        </interface>
    </hal>
    <hal format="hidl">
        <name>android.hardware.graphics.mapper</name>
        <transport arch="32">passthrough</transport>
        <version>2.0</version>
        <interface>
            <name>IMapper</name>
            <instance>default</instance>
        </interface>
    </hal>
```
There is no error in ```native_handle_clone```. I don't know if it is called or not. But the allocation is still failed with mysteries hidl callback transaction error.

remilia15, 2018-07-25:
wait why `**hwbinder**`? Are you highlighting or sth?

ngoquang2708, 2018-07-25:
Yeah, highlighting.

remilia15, 2018-07-25:
Have some logs line to debug it. I'm sure there will be errors, whatever you set passthrough or binderized

Stricted, 2018-08-02:
hi guys

i have the same problem on my device (mediatek soc)
after digging around abit i found that it fails here https://github.com/LineageOS/android_hardware_interfaces/blob/lineage-15.1/graphics/mapper/2.0/default/GrallocMapper.cpp#L137
as you also found out native_handle_clone fails
i liked your idea in passing the -1 fd to mali
i was actually successful with that, i just copied native_handle_clone and modified it (you can actually do that in GrallocMapper.cpp which would allow us to just build a custom interface)
```
native_handle_t* native_handle_clone_mt8127(const native_handle_t* handle) {
    native_handle_t* clone = native_handle_create(handle->numFds, handle->numInts);
    if (clone == NULL) return NULL;

    for (int i = 0; i < handle->numFds; i++) {
        clone->data[i] = handle->data[i];
    }

    memcpy(&clone->data[handle->numFds], &handle->data[handle->numFds],
           sizeof(int) * handle->numInts);

    return clone;
}
```
well that kinda worked but now the gralloc goes hell with 
`E [MALI][Gralloc-ERROR]: int gralloc_register_buffer(const gralloc_module_t*, buffer_handle_t):110 Can't register buffer 0xab7a3320 as it is a framebuffer` and it hits this check afterwards https://github.com/LineageOS/android_hardware_interfaces/blob/lineage-15.1/graphics/mapper/2.0/default/GrallocMapper.cpp#L144

thats where im stuck now

remilia15, 2018-08-02:
First of all, you can never clone FB's shared fd, because its fd is always negative

Second, even if you force it to clone, mali won't recognize the cloned data, because there's no FD to read and access, result in mali fails to allocate physical address for FB (tho this is done by actually read the FB fd instead of the shared fd through clone process, and mali can never do it with HIDL mapper)

Third, from the 2nd reason, it's also result in FB fd is being treated as shared_fd, result in theh given error line in logcat as you mentioned

ngoquang2708, 2018-08-14:
```
08-14 20:19:50.950  2636  2674 D vndksupport: Loading /vendor/lib/hw/gralloc.scx15.so from current namespace instead of sphal namespace.
```
Will this cause any issue? If this is the case, then the initialized framebuffer memory should have to be re-map again?

remilia15, 2018-08-14:
Just ignore it. We don't support VNDK libraries

ngoquang2708, 2018-08-16:
@remilia15 What do you mean by support VNDK libraries? Is it using only VNDK libraries for hardware modules?

remilia15, 2018-08-16:
The actual fact is that we don't have VNDK blobs, rather than not supporting VNDK libraries
VNDK libs are used for treble configuration. In short, everything in vendor can only rely on core Android libraries (libc, libc++, ....), and not anything else, or it would break VNDK compliance

ngoquang2708, 2018-08-19:
I have observed something link this, if I remove [this](https://github.com/ngoquang2708/android_hardware_sprd/blob/lineage-15.1/gralloc/scx15/alloc_device.cpp#L307) line and modify the ```native_handle_clone``` to allow dupplication of negative file descriptor like what @Stricted did, then ```libMali``` can successfully map the physical memory of that native handle and no crashing of ```surfaceflinger``` but nothing happen in the display.
What I did is based on what QCOM did with their ```gpu_context_t::free_impl``` function in their ```libgralloc0```.
Based on what I have observed, I found out that after successfully cloned the framebuffer native handle, the ```GraphicBuffer``` in ```libui``` then calling free on the original handle, cause ```libMali``` to crash since I think it is still hold the old handle because there is no ```native_handle_clone``` to mess with until N.

@Stricted You can modify your ```gralloc_register_buffer``` function to return 0 if the buffer_handle_t is a framebuffer handle like [this](https://github.com/ngoquang2708/android_hardware_sprd/blob/lineage-15.1/gralloc/scx15/mapper.cpp#L51) to see that is the difference.

remilia15, 2018-08-19:
Mali uses ION fd with default ION implementation, that's why we have to close fd