ngoquang2708/android_device_samsung_vivalto3gvn #101: Cannot wake the device up

ngoquang2708, 2018-07-07:


remilia15, 2018-07-07:
Needs reverting autosleep commit from system_core, or update kernel (idk about this)
https://github.com/LineageOS/android_system_core/commit/021d60be98ca2eb72b2c63e3b1fea2169b7c2b69#diff-c48ffca6d3844748d080c12a9a312012

ngoquang2708, 2018-07-07:
I set the sleep time to 30min :)

ngoquang2708, 2018-07-07:
Have you received my email?

remilia15, 2018-07-07:
Now I noticed this XD

ngoquang2708, 2018-07-07:
Oh, don't read email guy :)
Anyway. Did the new rild crash?

ngoquang2708, 2018-07-07:
Or it is running normally but you cannot make call?

remilia15, 2018-07-07:
rild didn't crash, at least if you don't use multisim's `RIL_onUnsolicitedResponse`, because it has 4 arguments and our libreference-ril_sp has 3 arguments, causing segfaults

remilia15, 2018-07-07:
Yeah I couldn't make calls. Somehow on dialing function or when it sends `AT+SSAM=0` command to modem, the RIL got crashed

ngoquang2708, 2018-07-07:
Getting rild up and running is a big success here for you. Mine always crash. I have a fear that LD_PRELOAD cause it. I will try to shim it instead.

remilia15, 2018-07-07:
That was just a little bit disassembling and understanding pseudo-C codes using IDA Pro XD

What I am trying to understand is why Sammy uses such big size for its `RIL_CardStatus_v6` struct. AOSP has 280 but my stock blob has 440. Inserting ints to that struct in AOSP isn't a good idea

ngoquang2708, 2018-07-07:
LD_PRELOAD some how cash my rild (aosp ril from your repo) :D. Now it can run with shim. So I have to modify my blobs and my shim :).

remilia15, 2018-07-07:
So you got signals right?

ngoquang2708, 2018-07-07:
Nothing yet. rild just not crash.

remilia15, 2018-07-07:
Maybe I know why it crashes. `property_get/set` args have been modified in O

ngoquang2708, 2018-07-07:
What changes?

remilia15, 2018-07-07:
Oh damn forget it. I used GetProperty() for my init instead of property_get() XD

remilia15, 2018-07-07:
Just checked, stuffs didn't change

ngoquang2708, 2018-07-07:
Haha. Same stuff for C, C++, Java. We all got confused. Isn't it!

ngoquang2708, 2018-07-07:
OMG. I got signal :o

remilia15, 2018-07-07:
So you're surprised like me XD

remilia15, 2018-07-07:
I guess your RIL bugs now are exactly the same as mine

ngoquang2708, 2018-07-07:
Yeah. No call. But USSD :D. Both SIM.
Se do you know where we should make changes in Java?
Previously we have used BOAD_RIL_CLASS to make changes. Now it seems to be gone.

remilia15, 2018-07-07:
That `BOARD_RIL_CLASS` is ditched in O. We all have to rebuild libril and since, most of java changes are just for that prebuilt lib so, if we were doing changes for RIL class before, do the same for libril at present

To be honest, sprd RIL is AOSP-friendly like other Samsung devices. Just because we were lazy to get it working though, or else it's very easy to bring up stuffs as for now XD