ngoquang2708/android_device_samsung_vivalto3gvn #102: Cannot make calls with  open source rild & libril

ngoquang2708, 2018-07-08:


ngoquang2708, 2018-07-08:
Commits https://github.com/ngoquang2708/android_device_samsung_vivalto3gvn/commit/8eb6cbaf0be745211b270dd440127d95e8784d9e and https://github.com/ngoquang2708/android_hardware_sprd/commit/b41febe292acca5f6df6f3d7cf974912126eeec7 seem to fix this issue.
@remilia15 I can make a call now. Want your confirmation.

remilia15, 2018-07-08:
I knew it. There must be a NULL in CallDetails
I'll test your commit now

remilia15, 2018-07-08:
Btw how did you figure out this issue?

ngoquang2708, 2018-07-08:
IDA is such a great tools. Why haven't I discover it earlier :)

remilia15, 2018-07-08:
Well it's a great tool for disassembling C libraries, but with my noob level, understanding those assembled codes is a problem

A nice thing is that I didn't even touch dial function in stock libril o.O
Logcat just give serviceDied upon dialing and sending commands to modem, nothing else. Maybe because I use libsecril-shim

ngoquang2708, 2018-07-08:
I have no knowledge of assembly either. Just guessing. Looking for old thing that we did in Java to make RIL work. Then re-apply it in C.

remilia15, 2018-07-08:
Confirmed working

remilia15, 2018-07-08:
Just ditched dial function in libsecril-shim. Now calls work immediately

remilia15, 2018-07-08:
Actually I use libsecril-shim for IMEI. Are your IMEIs in Settings unknown?

ngoquang2708, 2018-07-08:
Yeah. All are unknown :)

remilia15, 2018-07-08:
First is that IMEI is unknown
Second, signal bars will never change
Third, if you got IMEI working, current static radio impl will derp our 3G RAF (if you go to `Settings` -> `Mobile network` then `Advanced`), returning `-1` as RAF value and throw `RIL_E_GENERIC_FAILURE`

These are what I've found so far

ngoquang2708, 2018-07-08:
Do you know what is the ```-c``` parameter to ```rild```? What its value means? I have changed it to like 0 then 1 then 2 and got no signal. 1 2 3 still no signal. But left it as is as your as 0 2 3 work. Confuse me like the way old rild socket!

remilia15, 2018-07-08:
It's the same thing as we did for rild sockets
`-c` para stands for clientId, which is renamed from rild in O

ngoquang2708, 2018-07-08:
Ok. Wonder why it behaves like old one :)

closed 2018-07-11 by ngoquang2708