ngoquang2708/android_device_samsung_vivalto3gvn #103: reference-ril send some strange onRequestComplete messages

ngoquang2708, 2018-07-09:
Commit https://github.com/ngoquang2708/android_hardware_sprd/commit/f8e61e928294a74b34c4b6d45fba73431028044f address this issue.
Radio logcat:
```
07-09 22:49:20.516   167   336 V RILClient: [*] RecordReqHistory(): token(1), ID(59)
07-09 22:49:20.516   167   336 V RILClient: [*] RecordReqHistory(): token(1), ID(59)
07-09 22:49:20.516   167   336 V RILClient: SendOemRequestHookRaw(): token = 1
07-09 22:49:20.516   167   336 V RILClient: [*] FindReqHandler(): token(1)
07-09 22:49:20.517   167   336 V RILClient: [*] ClearReqHistory(): token(1)
07-09 22:49:20.524   191   391 E secril-shim: pRI->pCI is invalid: 14
07-09 22:49:20.524   191   391 E RILC    : RIL_onRequestComplete: invalid RIL_Token
07-09 22:49:20.525   167  1702 V RILClient: processRxBuffer: status 0 response_type 0
07-09 22:49:20.525   167  1702 V RILClient: processSolicited()
07-09 22:49:20.525   167  1702 E RILClient: processSolicited: Invalid Token
07-09 22:49:20.525   167  1702 E RILClient: RxReaderFunc: processRXBuffer returns 3
07-09 22:49:20.525   167  1702 V RILClient: [*] RxReaderFunc() b_connect=1
```

remilia15, 2018-07-09:
I guess RIL will crash after this

ngoquang2708, 2018-07-09:
Verry true.

ngoquang2708, 2018-07-09:
```libril``` impl of ```onRequestComplete``` has a guard check on invalid token value. Unfortunately, our shim do not have necessarily requirement to perform that check so I just ignore it and let ```libril``` handle it. We do not have interest on these response though.

remilia15, 2018-07-09:
There must be something that sends and invokes `RIL_REQUEST_OEM_HOOK_RAW`, and that thing is a request. Could you try to dig the full logcat or send it here so I can have a look at it?

This is like `RIL_REQUEST_LAST_CALL_FAIL_CAUSE`'s responseLen bug, which also throws some invalid pointer and kills the RIL, so I had to revert its shim which I removed before

remilia15, 2018-07-09:
Yes, exactly

remilia15, 2018-07-09:
Btw after this, could you please look at this: "Go to Settings -> Mobile network -> Press on `Advanced`, RIL will crash after that if your preferred network mode is 3G"

The reason I thought was the RAF value being invalid from logcat

ngoquang2708, 2018-07-09:
Confirmed. It is not happen if I am not use shim before. There is another thing. The Network field is set to sim one carrier on both sim.

remilia15, 2018-07-09:
You didn't face it before because IMEI was NULL, and you can't even change your preferred network type. It'll stay on HSPA or EDGE forever

And yes, the second thing is confirmed