ngoquang2708/android_device_samsung_vivalto3gvn #108: Cannot play video

ngoquang2708, 2018-07-11:


remilia15, 2018-07-11:
It's fine playing here
Do you have logs?

ngoquang2708, 2018-07-11:
Its here: https://gist.github.com/ngoquang2708/a6777ad7464c8ca1106470621a707f0f.
Do I need to update xml config?

remilia15, 2018-07-11:
I think there's an unsupported codec in your configs
`07-11 21:20:05.037   184  1781 E ACodec  : Unable to instantiate a decoder for type 'audio/mp4a-latm' with err 0xfffffffe.`

remilia15, 2018-07-11:
Try to have a check on boot, see whether sprd OMX codecs are initialized or not

ngoquang2708, 2018-07-11:
Wait. Did ffmpeg removed?

remilia15, 2018-07-11:
Maybe. I can't see it in my system

ngoquang2708, 2018-07-11:
Aghh. I have included it in my ```media_codecs.xml```. That why codecs is failed to init. I have just removed it. Music is working now. Still no video though.

ngoquang2708, 2018-07-11:
New logcat after removed ffmpeg xml: https://gist.github.com/ngoquang2708/2d410d8a7a1d146982cf69c1d9592620.

remilia15, 2018-07-11:
First is that why it's using SW decoder instead of HW one 😕 
Second, you need to disable treble OMX: `persist.media.treble_omx=false`

ngoquang2708, 2018-07-11:
You have go all through this, right :D

remilia15, 2018-07-11:
Yeah, since first LOS 15.0 releases :v

ngoquang2708, 2018-07-12:
Fixed by https://github.com/ngoquang2708/android_device_samsung_vivalto3gvn/commit/f8cf92ac8910b0ba830bae1273b463e5280239c6 and https://github.com/ngoquang2708/android_device_samsung_vivalto3gvn/commit/f5da1991a0f396c51b67c0061db113fd4b6c7aa9.

closed 2018-07-12 by ngoquang2708