ngoquang2708/android_device_samsung_vivalto3gvn #109: WiFi hotspot does not work

ngoquang2708, 2018-07-11:


ngoquang2708, 2018-07-22:
@remilia15 I have noticed something. I previous version of Android. When we turn on WiFi hotspot, hostapd change the firmware path.

ngoquang2708, 2018-07-22:
@remilia15 Here it goes!

remilia15, 2018-07-22:
Nice. My kernel had the commit you mentioned already

ngoquang2708, 2018-07-22:
OK. About the firmware path change, have you checked?

remilia15, 2018-07-22:
No, I haven't. What do you have in logcat?

remilia15, 2018-07-22:
I'm think about where does this commit goes
https://github.com/LineageOS/android_hardware_libhardware_legacy/commit/1650ddc3f6c0e8e29f8a5639f226358998f49738

remilia15, 2018-07-22:
Oh it's here
https://github.com/LineageOS/android_frameworks_opt_net_wifi/tree/lineage-15.1/libwifi_hal

ngoquang2708, 2018-07-22:
Nothing interesting in logcat. But kmsg shows that hostapd start with the same firmware as wpa_supplicant, not the one that for AP mode.
```
  271.766941] init: starting service 'hostapd'...
[  271.950538] \x0aDongle Host Driver, version 1.141.64 (r)\x0aCompiled in drivers/net/wireless/bcmdhd on Jul 20 2018 at 19:11:12
[  271.950657] wl_android_wifi_on in
[  271.950671] wifi_platform_set_power = 1
[  271.950680] wlan_device_power:1 
[  272.193183] sdio_reset_comm():
[  272.208106] F1 signature OK, socitype:0x1 chip:0xa9a6 rev:0x0 pkg:0x3
[  272.208829] DHD: dongle ram size is set to 524288(orig 524288) at 0x0
[  272.209674] REVISION SPECIFIC feature is not required
[  272.209694] dhdsdio_download_firmware: firmware path=/vendor/firmware/fw_bcmdhd.bin, nvram path=/vendor/etc/wifi/nvram_net.txt
[  272.262660] sdioh_request_buffer: [1] doing memory copy buf=c1966000, len=1527
[  272.264976] dhdsdio_write_vars: Download, Upload and compare of NVRAM succeeded.

```

ngoquang2708, 2018-07-29:
I see that netd's SoftApController is removed.

ngoquang2708, 2018-07-29:
It is used to switch the firmware between AP and STA mode.