ngoquang2708/android_device_samsung_vivalto3gvn #117: IptablesRestoreController failed

ngoquang2708, 2018-07-22:
Log: https://gist.github.com/ngoquang2708/190e863a7a2700459028e92102bf1ef5.

ngoquang2708, 2018-07-22:
All tethering issues seem relate to this issue.

remilia15, 2018-07-22:
You need these:
CONFIG_IP_NF_MATCH_RPFILTER=y
CONFIG_IP6_NF_MATCH_RPFILTER=y

ngoquang2708, 2018-07-22:
Hope it works.

remilia15, 2018-07-22:
This is the solution that works since N. On O idk what about old bcm4343 kernel, but mine doing weird stuffs. WiFi Tethering just work as placebo (it can be turned on, but give no hotspot signals). Maybe some kernel-related commits needed again?

remilia15, 2018-07-22:
If it shows error, then these 2 configs enabled will do their jobs, but if it doesn't show error (just keep turning on only, like my situation) then it's a different thing

ngoquang2708, 2018-07-22:
OK. Guess what, I am in the same situation as you now.
But now bluetooth tethering works!

remilia15, 2018-07-22:
USB tethering will work also. Tested on my W10 PC

ngoquang2708, 2018-07-22:
My USB seems not to work like expected. No MTP, so no tethering.

ngoquang2708, 2018-07-22:
IptablesRestoreController failure is still appear after reboot. But not when enable or disable tethering.

ngoquang2708, 2018-07-22:
So I think the invisible SSID is related to hostapd.

ngoquang2708, 2018-07-22:
Fixed by commit https://github.com/ngoquang2708/android_kernel_samsung_vivalto3gvn/commit/39d9f1fdf48e145866aee0fbf7a950edb89a8aee.

closed 2018-07-22 by ngoquang2708

remilia15, 2018-07-22:
LOL just about to comment on the issue you pinged me